simondankelmann/Bluetooth-LE-Spam #2: Immediately stops advertising

yoshimo, 2023-11-01:
I just tried both APKs you have released.
On a MIUI device with Android 10 the app requests location permissions for the time the app is in the foreground.
So far so good, but once you press the start button it immediately stops again without any line telling me that it attempted to send out a request.

simondankelmann, 2023-11-01:
Hi, could you run the App in Android Studio and provide the Error Message here ?

loss-and-quick, 2023-11-01:
> I just tried both APKs you have released. On a MIUI device with Android 10 the app requests location permissions for the time the app is in the foreground. So far so good, but once you press the [...]

Same on `EMUI`

ErfanDL, 2023-11-01:
same problem on Galaxy A70

Driv28, 2023-11-02:
Same problem with Version 1.0. Phone: Xiaomi Redmi K20 - Android 11 - Snapdragon 730.

Works with Galaxy S20 tho

ha2ked, 2023-11-02:
Same on the s8

ha2ked, 2023-11-02:
I do not know if this is the right info but this is the log i got 

2023-11-02 12:36:46.789 13993-13993 ViewRootIm...nActivity] de.simon.dankelmann.bluetoothlespam  D  ViewPostIme pointer 0
2023-11-02 12:36:46.920 13993-13993 ViewRootIm...nActivity] de.simon.dankelmann.bluetoothlespam  D  ViewPostIme pointer 1
2023-11-02 12:36:46.937 13993-13993 BluetoothAdapter        de.simon.dankelmann.bluetoothlespam  D  STATE_ON
2023-11-02 12:36:46.938 13993-13993 BluetoothAdapter        de.simon.dankelmann.bluetoothlespam  D  STATE_ON
2023-11-02 12:36:46.939 13993-13993 BluetoothAdapter        de.simon.dankelmann.bluetoothlespam  D  STATE_ON
2023-11-02 12:36:46.941 13993-13993 Advertisem...oopService de.simon.dankelmann.bluetoothlespam  D  Hardware Check returns: true
2023-11-02 12:36:47.014 13993-13993 PermissionCheck         de.simon.dankelmann.bluetoothlespam  D  Permission not granted: android.permission.BLUETOOTH_ADVERTISE
2023-11-02 12:36:47.014 13993-13993 BluetoothL...entService de.simon.dankelmann.bluetoothlespam  D  Missing permission to execute advertisement
2023-11-02 12:36:47.014 13993-13993 Advertisem...oopService de.simon.dankelmann.bluetoothlespam  D  Added advertiser for: Beats Studio Buds
2023-11-02 12:36:47.023 13993-13993 Activity                de.simon.dankelmann.bluetoothlespam  W  Can request only one set of permissions at a time
2023-11-02 12:36:47.023 13993-13993 PermissionCheck         de.simon.dankelmann.bluetoothlespam  D  Permission not granted: android.permission.BLUETOOTH_ADVERTISE
2023-11-02 12:36:47.023 13993-13993 BluetoothL...entService de.simon.dankelmann.bluetoothlespam  D  Missing permission to stop advertisement
2023-11-02 12:36:47.024 13993-13993 Activity                de.simon.dankelmann.bluetoothlespam  W  Can request only one set of permissions at a time
2023-11-02 12:36:47.024 13993-13993 PermissionCheck         de.simon.dankelmann.bluetoothlespam  D  Permission not granted: android.permission.BLUETOOTH_ADVERTISE
2023-11-02 12:36:47.024 13993-13993 BluetoothL...entService de.simon.dankelmann.bluetoothlespam  D  Missing permission to stop advertisement
2023-11-02 12:36:47.024 13993-13993 Activity                de.simon.dankelmann.bluetoothlespam  W  Can request only one set of permissions at a time
2023-11-02 12:36:47.025 13993-13993 PermissionCheck         de.simon.dankelmann.bluetoothlespam  D  Permission not granted: android.permission.BLUETOOTH_ADVERTISE
2023-11-02 12:36:47.025 13993-13993 BluetoothL...entService de.simon.dankelmann.bluetoothlespam  D  Missing permission to stop advertisement
2023-11-02 12:36:47.025 13993-13993 Activity                de.simon.dankelmann.bluetoothlespam  W  Can request only one set of permissions at a time
2023-11-02 12:36:47.025 13993-13993 PermissionCheck         de.simon.dankelmann.bluetoothlespam  D  Permission not granted: android.permission.BLUETOOTH_ADVERTISE
2023-11-02 12:36:47.025 13993-13993 BluetoothL...entService de.simon.dankelmann.bluetoothlespam  D  Missing permission to stop advertisement
2023-11-02 12:36:47.025 13993-13993 Activity                de.simon.dankelmann.bluetoothlespam  W  Can request only one set of permissions at a time
2023-11-02 12:36:47.025 13993-13993 PermissionCheck         de.simon.dankelmann.bluetoothlespam  D  Permission not granted: android.permission.BLUETOOTH_ADVERTISE
2023-11-02 12:36:47.025 13993-13993 BluetoothL...entService de.simon.dankelmann.bluetoothlespam  D  Missing permission to stop advertisement
2023-11-02 12:36:47.026 13993-13993 Activity                de.simon.dankelmann.bluetoothlespam  W  Can request only one set of permissions at a time
2023-11-02 12:36:47.026 13993-13993 PermissionCheck         de.simon.dankelmann.bluetoothlespam  D  Permission not granted: android.permission.BLUETOOTH_ADVERTISE
2023-11-02 12:36:47.026 13993-13993 BluetoothL...entService de.simon.dankelmann.bluetoothlespam  D  Missing permission to stop advertisement
2023-11-02 12:36:47.026 13993-13993 Activity                de.simon.dankelmann.bluetoothlespam  W  Can request only one set of permissions at a time
2023-11-02 12:36:47.026 13993-13993 PermissionCheck         de.simon.dankelmann.bluetoothlespam  D  Permission not granted: android.permission.BLUETOOTH_ADVERTISE
2023-11-02 12:36:47.026 13993-13993 BluetoothL...entService de.simon.dankelmann.bluetoothlespam  D  Missing permission to stop advertisement
2023-11-02 12:36:47.027 13993-13993 Activity                de.simon.dankelmann.bluetoothlespam  W  Can request only one set of permissions at a time
2023-11-02 12:36:47.027 13993-13993 PermissionCheck         de.simon.dankelmann.bluetoothlespam  D  Permission not granted: android.permission.BLUETOOTH_ADVERTISE
2023-11-02 12:36:47.027 13993-13993 BluetoothL...entService de.simon.dankelmann.bluetoothlespam  D  Missing permission to stop advertisement
2023-11-02 12:36:47.028 13993-13993 Activity                de.simon.dankelmann.bluetoothlespam  W  Can request only one set of permissions at a time
2023-11-02 12:36:47.028 13993-13993 PermissionCheck         de.simon.dankelmann.bluetoothlespam  D  Permission not granted: android.permission.BLUETOOTH_ADVERTISE
2023-11-02 12:36:47.028 13993-13993 BluetoothL...entService de.simon.dankelmann.bluetoothlespam  D  Missing permission to stop advertisement
2023-11-02 12:36:47.028 13993-13993 Activity                de.simon.dankelmann.bluetoothlespam  W  Can request only one set of permissions at a time
[... cut ...]
2023-11-02 12:36:47.271 13993-13993 InputMethodManager      de.simon.dankelmann.bluetoothlespam  D  prepareNavigationBarInfo() DecorView@d33a28b[MainActivity]
2023-11-02 12:36:47.272 13993-13993 InputMethodManager      de.simon.dankelmann.bluetoothlespam  D  getNavigationBarColor() -16381933
2023-11-02 12:36:47.293 13993-13993 ViewRootIm...nActivity] de.simon.dankelmann.bluetoothlespam  D  Relayout returned: old=[0,0][1080,2220] new=[0,0][1080,2220] result=0x1 surface={valid=true 503279755264} changed=false

ha2ked, 2023-11-02:
So my conclusion is that the android.permission.BLUETOOTH_ADVERTISE permission does not exist on android nine or lower

Driv28, 2023-11-02:
> So my conclusion is that the android.permission.BLUETOOTH_ADVERTISE permission does not exist on android nine or lower

I can try to take a log tomorrow but the OP has Android 10 and my phone is on Android 11

simondankelmann, 2023-11-02:
I changed the used permissions in the Manifest.xml accordingly to: 

https://developer.android.com/develop/connectivity/bluetooth/bt-permissions

please try again and let us know if it helped :)

ha2ked, 2023-11-02:
Tried on google pixel 7 running android 9 and kept crashing this is the log

2023-11-02 17:08:22.139  4723-4723  bluetoothlespa          de.simon.dankelmann.bluetoothlespam  W  Accessing hidden field Landroid/os/Trace;->TRACE_TAG_APP:J (light greylist, reflection)
2023-11-02 17:08:22.140  4723-4723  bluetoothlespa          de.simon.dankelmann.bluetoothlespam  W  Accessing hidden method Landroid/os/Trace;->isTagEnabled(J)Z (light greylist, reflection)
2023-11-02 17:08:22.200  4723-4723  AppCompatDelegate       de.simon.dankelmann.bluetoothlespam  D  Checking for metadata for AppLocalesMetadataHolderService : Service not found
2023-11-02 17:08:22.674  4723-4723  bluetoothlespa          de.simon.dankelmann.bluetoothlespam  W  Accessing hidden method Landroid/view/View;->computeFitSystemWindows(Landroid/graphics/Rect;Landroid/graphics/Rect;)Z (light greylist, reflection)
2023-11-02 17:08:23.066  4723-4723  BluetoothAdapter        de.simon.dankelmann.bluetoothlespam  E  Bluetooth binder is null
2023-11-02 17:08:23.170  4723-4723  OpenGLRenderer          de.simon.dankelmann.bluetoothlespam  D  HWUI GL Pipeline
2023-11-02 17:08:23.231  4723-4723  bluetoothlespa          de.simon.dankelmann.bluetoothlespam  W  Accessing hidden method Landroid/view/ViewGroup;->makeOptionalFitsSystemWindows()V (light greylist, reflection)
2023-11-02 17:08:23.274  4723-4723  AndroidRuntime          de.simon.dankelmann.bluetoothlespam  D  Shutting down VM
2023-11-02 17:08:23.276  4723-4723  AndroidRuntime          de.simon.dankelmann.bluetoothlespam  E  FATAL EXCEPTION: main
                                                                                                    Process: de.simon.dankelmann.bluetoothlespam, PID: 4723
                                                                                                    java.lang.RuntimeException: Unable to resume activity {de.simon.dankelmann.bluetoothlespam/de.simon.dankelmann.bluetoothlespam.MainActivity}: java.lang.NullPointerException: getValue(...) must not be null
                                                                                                    	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3784)
                                                                                                    	at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:3816)
                                                                                                    	at android.app.servertransaction.ResumeActivityItem.execute(ResumeActivityItem.java:51)
                                                                                                    	at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:145)
                                                                                                    	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:70)
                                                                                                    	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1808)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:106)
                                                                                                    	at android.os.Looper.loop(Looper.java:193)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:6669)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)
                                                                                                    Caused by: java.lang.NullPointerException: getValue(...) must not be null
                                                                                                    	at de.simon.dankelmann.bluetoothlespam.MainActivity.getBluetoothAdapter(MainActivity.kt:60)
                                                                                                    	at de.simon.dankelmann.bluetoothlespam.MainActivity.onResume(MainActivity.kt:81)
                                                                                                    	at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1412)
                                                                                                    	at android.app.Activity.performResume(Activity.java:7292)
                                                                                                    	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3776)
                                                                                                    	at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:3816) 
[... cut ...]
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method) 
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493) 
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)

Driv28, 2023-11-02:
> I changed the used permissions in the Manifest.xml accordingly to:
> 
> https://developer.android.com/develop/connectivity/bluetooth/bt-permissions
> 
> please try again and let us know if it helped :)

Thank you for your fast reply. Tested both apks. I now get a warning when I install the app but same error as before.

ha2ked, 2023-11-02:
Yup same here

ha2ked, 2023-11-02:
> Thank you for your fast reply. Tested both apks. I now get a warning when I install the app but same error as before.

Same here

simondankelmann, 2023-11-02:
Once more i changed the way permissions are requested at runtime due to this line in the Log: 

`Can request only one set of permissions at a time`

please check out:

https://github.com/simondankelmann/Bluetooth-LE-Spam/releases/tag/v0.0.2

ha2ked, 2023-11-02:
> Once more i changed the way permissions are requested at runtime due to this line in the Log: 
> 
> `Can request only one set of permissions at a time`
> 
> please check out:
> 
> https://github.com/simondankelmann/Bluetooth-LE-Spam/releases/tag/v0.0.2

Sorry ain't working will comment log in a sec

ha2ked, 2023-11-02:
Here is log from latest apk

2023-11-02 18:23:31.493 28178-28178 ViewRootIm...nActivity] de.simon.dankelmann.bluetoothlespam  D  ViewPostIme pointer 0
2023-11-02 18:23:31.541 28178-28178 ViewRootIm...nActivity] de.simon.dankelmann.bluetoothlespam  D  ViewPostIme pointer 1
2023-11-02 18:23:31.579 28178-28178 BluetoothAdapter        de.simon.dankelmann.bluetoothlespam  D  STATE_ON
2023-11-02 18:23:31.580 28178-28178 BluetoothAdapter        de.simon.dankelmann.bluetoothlespam  D  STATE_ON
2023-11-02 18:23:31.582 28178-28178 BluetoothAdapter        de.simon.dankelmann.bluetoothlespam  D  STATE_ON
2023-11-02 18:23:31.587 28178-28178 Advertisem...oopService de.simon.dankelmann.bluetoothlespam  D  Hardware Check returns: true
2023-11-02 18:23:31.645 28178-28178 PermissionCheck         de.simon.dankelmann.bluetoothlespam  D  Permission not granted: android.permission.BLUETOOTH_ADVERTISE
2023-11-02 18:23:31.645 28178-28178 BluetoothL...entService de.simon.dankelmann.bluetoothlespam  D  Missing permission to execute advertisement
2023-11-02 18:23:31.645 28178-28178 Advertisem...oopService de.simon.dankelmann.bluetoothlespam  D  Added advertiser for: SRS-XB43
2023-11-02 18:23:31.666 28178-28178 Activity                de.simon.dankelmann.bluetoothlespam  W  Can request only one set of permissions at a time
2023-11-02 18:23:31.667 28178-28178 PermissionCheck         de.simon.dankelmann.bluetoothlespam  D  Permission not granted: android.permission.BLUETOOTH_ADVERTISE
2023-11-02 18:23:31.667 28178-28178 BluetoothL...entService de.simon.dankelmann.bluetoothlespam  D  Missing permission to stop advertisement
2023-11-02 18:23:31.667 28178-28178 Activity                de.simon.dankelmann.bluetoothlespam  W  Can request only one set of permissions at a time
2023-11-02 18:23:31.667 28178-28178 PermissionCheck         de.simon.dankelmann.bluetoothlespam  D  Permission not granted: android.permission.BLUETOOTH_ADVERTISE
2023-11-02 18:23:31.667 28178-28178 BluetoothL...entService de.simon.dankelmann.bluetoothlespam  D  Missing permission to stop advertisement
2023-11-02 18:23:31.668 28178-28178 Activity                de.simon.dankelmann.bluetoothlespam  W  Can request only one set of permissions at a time
2023-11-02 18:23:31.668 28178-28178 PermissionCheck         de.simon.dankelmann.bluetoothlespam  D  Permission not granted: android.permission.BLUETOOTH_ADVERTISE
2023-11-02 18:23:31.668 28178-28178 BluetoothL...entService de.simon.dankelmann.bluetoothlespam  D  Missing permission to stop advertisement
2023-11-02 18:23:31.669 28178-28178 Activity                de.simon.dankelmann.bluetoothlespam  W  Can request only one set of permissions at a time
2023-11-02 18:23:31.669 28178-28178 PermissionCheck         de.simon.dankelmann.bluetoothlespam  D  Permission not granted: android.permission.BLUETOOTH_ADVERTISE
2023-11-02 18:23:31.669 28178-28178 BluetoothL...entService de.simon.dankelmann.bluetoothlespam  D  Missing permission to stop advertisement
2023-11-02 18:23:31.669 28178-28178 Activity                de.simon.dankelmann.bluetoothlespam  W  Can request only one set of permissions at a time
2023-11-02 18:23:31.669 28178-28178 PermissionCheck         de.simon.dankelmann.bluetoothlespam  D  Permission not granted: android.permission.BLUETOOTH_ADVERTISE
2023-11-02 18:23:31.669 28178-28178 BluetoothL...entService de.simon.dankelmann.bluetoothlespam  D  Missing permission to stop advertisement
2023-11-02 18:23:31.670 28178-28178 Activity                de.simon.dankelmann.bluetoothlespam  W  Can request only one set of permissions at a time
2023-11-02 18:23:31.670 28178-28178 PermissionCheck         de.simon.dankelmann.bluetoothlespam  D  Permission not granted: android.permission.BLUETOOTH_ADVERTISE
2023-11-02 18:23:31.670 28178-28178 BluetoothL...entService de.simon.dankelmann.bluetoothlespam  D  Missing permission to stop advertisement
2023-11-02 18:23:31.671 28178-28178 Activity                de.simon.dankelmann.bluetoothlespam  W  Can request only one set of permissions at a time
2023-11-02 18:23:31.671 28178-28178 PermissionCheck         de.simon.dankelmann.bluetoothlespam  D  Permission not granted: android.permission.BLUETOOTH_ADVERTISE
2023-11-02 18:23:31.671 28178-28178 BluetoothL...entService de.simon.dankelmann.bluetoothlespam  D  Missing permission to stop advertisement
2023-11-02 18:23:31.672 28178-28178 Activity                de.simon.dankelmann.bluetoothlespam  W  Can request only one set of permissions at a time
2023-11-02 18:23:31.672 28178-28178 PermissionCheck         de.simon.dankelmann.bluetoothlespam  D  Permission not granted: android.permission.BLUETOOTH_ADVERTISE
2023-11-02 18:23:31.672 28178-28178 BluetoothL...entService de.simon.dankelmann.bluetoothlespam  D  Missing permission to stop advertisement
2023-11-02 18:23:31.672 28178-28178 Activity                de.simon.dankelmann.bluetoothlespam  W  Can request only one set of permissions at a time
2023-11-02 18:23:31.672 28178-28178 PermissionCheck         de.simon.dankelmann.bluetoothlespam  D  Permission not granted: android.permission.BLUETOOTH_ADVERTISE
2023-11-02 18:23:31.672 28178-28178 BluetoothL...entService de.simon.dankelmann.bluetoothlespam  D  Missing permission to stop advertisement
2023-11-02 18:23:31.673 28178-28178 Activity                de.simon.dankelmann.bluetoothlespam  W  Can request only one set of permissions at a time
[... cut ...]
2023-11-02 18:23:31.928 28178-28178 InputMethodManager      de.simon.dankelmann.bluetoothlespam  D  prepareNavigationBarInfo() DecorView@d33a28b[MainActivity]
2023-11-02 18:23:31.928 28178-28178 InputMethodManager      de.simon.dankelmann.bluetoothlespam  D  getNavigationBarColor() -16381933
2023-11-02 18:23:31.951 28178-28178 ViewRootIm...nActivity] de.simon.dankelmann.bluetoothlespam  D  Relayout returned: old=[0,0][1080,2220] new=[0,0][1080,2220] result=0x1 surface={valid=true 503280041984} changed=false
2023-11-02 18:24:10.511 28178-28178 ViewRootIm...nActivity] de.simon.dankelmann.bluetoothlespam  D  MSG_WINDOW_FOCUS_CHANGED 0 1

ha2ked, 2023-11-02:
Can you by any chance use a deferent permission than ADVERTISING

ha2ked, 2023-11-02:
In case this is any help this is the log on starting up the app 
[STARTUP-LOG.txt](https://github.com/simondankelmann/Bluetooth-LE-Spam/files/13243314/STARTUP-LOG.txt)

Driv28, 2023-11-02:
> > Once more i changed the way permissions are requested at runtime due to this line in the Log:
> > `Can request only one set of permissions at a time`
> > please check out:
> > https://github.com/simondankelmann/Bluetooth-LE-Spam/releases/tag/v0.0.2
> 
> Sorry ain't working will comment log in a sec

I confirm, same error

ha2ked, 2023-11-02:
Is this the problem this is in log

[image]

Geofferey, 2023-11-03:
`pm grant de.simon.dankelmann.bluetoothlespam  android.permission.BLUETOOTH_ADVERTISE’

`Exception occurred while executing 'grant':
java.lang.IllegalArgumentException: Unknown permission: android.permission.BLUETOOTH_ADVERTISE`

Wonder 🤔

simondankelmann, 2023-11-03:
Can you try granting all permissions manually in the app settings ?

Driv28, 2023-11-03:
> Can you try granting all permissions manually in the app settings ?

Yes, but No improvement

SpiritOTHawk, 2023-11-03:
Some issue on Galaxy A40 and Galaxy A3 (2017). Android 11 and 8.

simondankelmann, 2023-11-03:
Please try the latest Release:

https://github.com/simondankelmann/Bluetooth-LE-Spam/releases

the BLUETOOTH_ADVERTISE permission was introduced in API Level 31 (Android 12)  as @ha2ked mentioned, causing problems on older API Levels :

`Permission not granted: android.permission.BLUETOOTH_ADVERTISE`

Driv28, 2023-11-03:
> Please try the latest Release:
>
> https://github.com/simondankelmann/Bluetooth-LE-Spam/releases
>
> the BLUETOOTH_ADVERTISE permission was introduced in API Level 31 (Android 12) as @ha2ked mentioned, [...]

Yes works now. No more stopping ☺️☺️☺️

ha2ked, 2023-11-03:
> Yes works now. No more stopping ☺️☺️☺️

Yes working here thank you

closed 2023-11-03 by simondankelmann