simondankelmann/Bluetooth-LE-Spam #3: Immediately crash ater asking for permissions

AlexandreKueny, 2023-11-01:
I tried both apk on my Oneplus 9 pro with Android 13.
It asks for the permissions an then immediately crash. After that, the app won't start anymore.

vanwafang, 2023-11-01:
Same problem on my Galaxy S10 Lite, One UI version 5.1

simondankelmann, 2023-11-01:
Hi, could you run the App in Android Studio and provide the Error Message here ?

n00b69, 2023-11-01:
Enable Bluetooth before opening the app, it works then.

simondankelmann, 2023-11-01:
I added some null checks, it seems there was a problem when Bluetooth is turned off while you open the App. You might download the new Version and try again :) if the Error remains, it would be great if you could submit the Errormessage from Android Studio

vanwafang, 2023-11-01:
Thank you all. It works now  :)

AlexandreKueny, 2023-11-02:
I can confirm that the new apk is working fine for me also.
Thank you!

closed 2023-11-02 by AlexandreKueny

Yokthailand, 2023-11-02:
App stop after start advertising on s20fe android 13