simondankelmann/Bluetooth-LE-Spam #4: App crashes when include name is enabled

konstkarapan, 2023-11-01:
When enabling include name, the app crashes (pocox4pro5g)

simondankelmann, 2023-11-01:
Hi, could you run the App in Android Studio and provide the Error Message here ?

C10udburst, 2023-11-01:
```
FATAL EXCEPTION: main
Process: de.simon.dankelmann.bluetoothlespam, PID: 26321
java.lang.IllegalArgumentException: Legacy advertising data too big
	at android.bluetooth.le.BluetoothLeAdvertiser.startAdvertisingSet(BluetoothLeAdvertiser.java:386)
	at android.bluetooth.le.BluetoothLeAdvertiser.startAdvertisingSet(BluetoothLeAdvertiser.java:262)
	at de.simon.dankelmann.bluetoothlespam.Services.BluetoothLeAdvertisementService.startAdvertisingSet(BluetoothLeAdvertisementService.kt:121)
	at de.simon.dankelmann.bluetoothlespam.Services.AdvertisementLoopService.advertiseNextPackage(AdvertisementLoopService.kt:131)
	at de.simon.dankelmann.bluetoothlespam.Services.AdvertisementLoopService$getTimer$1.onTick(AdvertisementLoopService.kt:45)
	at android.os.CountDownTimer$1.handleMessage(CountDownTimer.java:145)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:7975)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:942)
```

simondankelmann, 2023-11-02:
Is it possible that you have a rather long bluetooth device name ?

konstkarapan, 2023-11-02:
You were right reducing my name's length solved the issue

closed 2023-11-02 by konstkarapan