simondankelmann/Bluetooth-LE-Spam #5: App does not work in the background

n00b69, 2023-11-01:
This is probably intentional, but it would be nice if background functionality were to be added.

simondankelmann, 2023-11-01:
Actually you are right, thats intentional at the moment. if you comment out the lines in the fragments onPause and onResume Method i think it will continue to work in the Background. But that could cause some weird behaviour when switching the Fragments.

bobeatsshit, 2023-11-01:
where would this be located if i wanted to enable this?

simondankelmann, 2023-11-02:
For Fast Pairing (Android): [here](https://github.com/simondankelmann/Bluetooth-LE-Spam/blob/main/app/src/main/java/de/simon/dankelmann/bluetoothlespam/ui/fastPairing/FastPairingFragment.kt)

For Swift Pairing (Windows): [here](https://github.com/simondankelmann/Bluetooth-LE-Spam/blob/main/app/src/main/java/de/simon/dankelmann/bluetoothlespam/ui/swiftPair/SwiftPairFragment.kt)

In this Method, the advertising is stopped:

```
    override fun onPause() {
        super.onPause()
        if(_advertisementLoopService != null){
            if(_advertisementLoopService!!.advertising){
                stopAdvertising()
            }
        }
    }
```

closed 2023-11-04 by simondankelmann