simondankelmann/Bluetooth-LE-Spam #15: App does not work in Huawei Devices

nmochea, 2023-11-02:
Immediately stops advertising

ha2ked, 2023-11-03:
Should hopefully work now try newest release

nmochea, 2023-11-04:
Unfortunately, when testing the new release version `1.0`, it forces stop after tapping the `Start Advertising` button.

DerGoogler, 2023-11-04:
@nmochea, To write a good issue you also should provide the information device, such as
- Android version 
- device model
- app version
- e.g. LogCat

nmochea, 2023-11-04:
@DerGoogler, Thanks for Enlightening
- Android version: `12`
- Device model: `MGA-LX9`
- App version: `1.0`

ha2ked, 2023-11-04:
> @DerGoogler, Thanks for Enlightening
> 
> * Android version: `12`
> * Device model: `MGA-LX9`
> * App version: `1.0`
> 
> Unfortunately, when testing the new release version `1.0`, it forces stop after tapping the `Start Advertising` button.

If I do say so myself the best described issue that I have seen so far anyway did you try the newsiest release and did that work

DerGoogler, 2023-11-04:
@nmochea I've found this link (https://consumer.huawei.com/en/support/content/en-us00895112/).I don't have experience with Huawei but **"maybe"** solves your issue.

ha2ked, 2023-11-04:
> @nmochea I've found this link (https://consumer.huawei.com/en/support/content/en-us00895112/).I don't have experience with Huawei but **"maybe"** solves your issue.

That link don`t work

nmochea, 2023-11-04:
> > @DerGoogler, Thanks for Enlightening
> > 
> > * Android version: `12`
> > * Device model: `MGA-LX9`
> > * App version: `1.0`
> > 
> > Unfortunately, when testing the new release version `1.0`, it forces stop after tapping the `Start Advertising` button.
> 
> If I do say so myself the best described issue that I have seen so far anyway did you try the newsiest release and did that work by the way we are on version [v0.0.3](https://github.com/simondankelmann/Bluetooth-LE-Spam/releases/tag/v0.0.3)

Testing `v0.0.3` still force stop the app.

DerGoogler, 2023-11-04:
@ha2ked sorry, my mistake while formatting. Here: https://consumer.huawei.com/en/support/content/en-us00895112/

ha2ked, 2023-11-04:
> > > @DerGoogler, Thanks for Enlightening
> > > 
> > > * Android version: `12`
> > > * Device model: `MGA-LX9`
> > > * App version: `1.0`
> > > 
> > > Unfortunately, when testing the new release version `1.0`, it forces stop after tapping the `Start Advertising` button.
> > 
> > 
> > If I do say so myself the best described issue that I have seen so far anyway did you try the newsiest release and did that work by the way we are on version [v0.0.3](https://github.com/simondankelmann/Bluetooth-LE-Spam/releases/tag/v0.0.3)
> 
> Testing version `v0.0.3` still force stop the app, after tapping the Start Advertising button.

Hmmm odd will try on virilized version of your machine

ha2ked, 2023-11-04:
> @ha2ked sorry, my mistake while formatting. Here: https://consumer.huawei.com/en/support/content/en-us00895112/

No prob

nmochea, 2023-11-04:
> Hmmm odd will try on virilized version of your machine

Regarding the issue of working on lower Android versions (e.g., 8) and specific models like the Huawei 2019, it's possible that the problem arises due to compatibility with higher Android versions. We hope to address this in future versions, possibly starting from version 12. Additionally, there seems to be an issue with the advertising button. I've also observed that the app does not request Bluetooth permission to turn on while in use, which can lead to the app freezing and not functioning properly.

ha2ked, 2023-11-04:
Could you please send us the log

simondankelmann, 2023-11-04:
Please check out this experimental version:

https://github.com/simondankelmann/Bluetooth-LE-Spam/blob/simondankelmann/experimentalAdvertising/app/debug/app-debug.apk

This needs some more testing but today i saw it working an a poco x3

simondankelmann, 2023-11-06:
@nmochea 

Please try again with the latest Version, go to Settings and enable "Use Advertising with Settings". Please also try  with both devices right next to each other because the range varies from device to device. If popups appear, please close the issue :)

nmochea, 2023-11-07:
Hi,
After testing the latest version, v1.0.2, I found that it works well on Huawei devices, ranging from Android version 8 to Android version 12. I'm pleased that it's finally working, and the implementation of Apple popups and modals is functioning correctly.

simondankelmann, 2023-11-07:
Great to hear that, thank your for testing and your feedback !

closed 2023-11-07 by simondankelmann