simondankelmann/Bluetooth-LE-Spam #18: Xiaomi 12 not working

byzuzayli, 2023-11-02:
Hi I also tried this app. I also tried from the nrf connect program, but I could not send any notifications. Are my phone's specs correct this way?  [image]

Justxd22, 2023-11-02:
Any updates on this? I test on xiaomi redmi note 10 pro same specs as screenshot (no le audio broadcast) , app working but no effect

simondankelmann, 2023-11-04:
Please check out this experimental version:

https://github.com/simondankelmann/Bluetooth-LE-Spam/blob/simondankelmann/experimentalAdvertising/app/debug/app-debug.apk

This needs some more testing but today i saw it working an a poco x3

simondankelmann, 2023-11-06:
@byzuzayli 

Please try again with the latest Version, go to Settings and enable "Use Advertising with Settings". Please also try  with both devices right next to each other because the range varies from device to device. If popups appear, please close the issue :)

Vitaliklidy, 2023-11-06:
> Hi I also tried this app. I also tried from the nrf connect program, but I could not send any notifications. Are my phone's specs correct this way? MIUI 14.0.6  [image]
> 
> 

I also tried in nrf connect but then I turn advertising tx power was changed from -7 to -11 dBm maybe that's a case

simondankelmann, 2023-11-13:
@byzuzayli did you succeed with any of the new releases ?

byzuzayli, 2023-11-16:
I bought flipper zero. I continue with this

closed 2023-11-16 by simondankelmann