simondankelmann/Bluetooth-LE-Spam #21: Please add more spam devices

RZXPRO1203, 2023-11-02:
Please add ios specific devices and samsung specific devices

RZXPRO1203, 2023-11-06:
Add samsung specific devices that work for samsung only (galaxy buds and all of that). I can give you the advertising data if you need, i want this added because the google fast pair feature doesnt work well on updated samsung phones. Samsung has their own feature for connecting to samsung devices which is vulnerable to spam

simondankelmann, 2023-11-06:
If you have that Data, please send it here, or create a pullrequest once there is some time i can take a look at that ;)

RZXPRO1203, 2023-11-06:
Unfortunately i only have the galaxy buds, but they work for spam

Company ID:
0075
Advertising:
420981029415032101885317012A063CE7EB000000001D00
Response:
006956721EA040366464000100

MemeCornucopia, 2023-11-17:
All the Samsung models are here
https://github.com/Flipper-XFW/Xtreme-Firmware/blob/dev/applications/external/ble_spam/protocols/easysetup.c

@simondankelmann

ha2ked, 2023-11-19:
There is many things in the works

simondankelmann, 2023-11-19:
Please check the latest Release:

https://github.com/simondankelmann/Bluetooth-LE-Spam/releases

Samsung Watches are added, Buds need more testing :)

closed 2023-11-19 by simondankelmann