simondankelmann/Bluetooth-LE-Spam #9: No pop-ups appearing on target device

herobrinenana, 2023-11-02:
For some reason I can't get any popups to appear on the target device? Is there a fix or is it supposed to not let me pair them?

n00b69, 2023-11-02:
I have not had success getting popups on my Galaxy S7 and Galaxy S10. A friend reported it not working on his S23 Ultra either.

Xiaomi phones work fine.

123321mario, 2023-11-02:
I can't get the app to send anything on my Pixel 7 Pro, will test more devices later today

Justxd22, 2023-11-02:
Same here with Xiaomi device (snapdragon chipset)

user4302, 2023-11-03:
How do you target a specific device?
I'm using the app on my android.
I tried the fast pairing as well as the swift pairing.
It just goes through devices, and I think it repeats the same devices.
None of my 2 other Android are getting spoofed
Am I doing it wrong?

ha2ked, 2023-11-03:
It really depends on host device and recipient device

simondankelmann, 2023-11-06:
Please try again with the latest Version, go to Settings and enable "Use Advertising with Settings". Please also try  with both devices right next to each other because the range varies from device to device. If popups appear, please close the issue :)

herobrinenana, 2023-11-06:
I tried using your suggestion, it's still not showing any pop-ups on Google fast pair compatible android devices. I attempted using Microsoft swift pair too and it didn't work. I'll attempt using a 3rd device later to check if it's the problem.

user4302, 2023-11-08:
I saw in an article that certain settings need to be disabled like nearby share and stuff like that.

(Apologies if you already mntioned this on the setup instructions)

But is that the case? Or should I be able to just install and run this with no issues?

simondankelmann, 2023-11-13:
@herobrinenana did you have any success ?

Justxd22, 2023-11-14:
@simondankelmann can confirm lastest build is working on my xiaomi device snapdragon

simondankelmann, 2023-11-15:
Closed for now as no more Information is provided

closed 2023-11-15 by simondankelmann