simondankelmann/Bluetooth-LE-Spam #23: It Starts on z Fold 3 but nothing happens

GammaPotato, 2023-11-03:
Newest app version but not working. Windows shows only on device "Unknown Device". Fast pairing didnt work on android devices.

ha2ked, 2023-11-05:
Nice pls close Issue
:)

GammaPotato, 2023-11-05:
> Nice pls close Issue :)

Uhm. Swift is not working. "Unknown device" on newest version

GammaPotato, 2023-11-07:
@simondankelmann 

Action "Modals' is working now on apple decives but pop didnt work. The app shows success full send but nothing happens. Holding the phones together will not help to

simondankelmann, 2023-11-08:
Which App version did you use ? Please try again after activating the Switch in the Settings. If you already had it activated please try without it.

simondankelmann, 2023-11-13:
Closed for now as there is no further Information

closed 2023-11-13 by simondankelmann