simondankelmann/Bluetooth-LE-Spam #28: Won't install on older android version

Pepe-454alt, 2023-11-03:
I was using samsung galaxy A3 (2016) _android 7.0_ if i try to instal the apk it won't i enabled unknown aplications. After i try downloading it reads: **While analyzing the package an error apeard.**

ha2ked, 2023-11-03:
Should work with newest release

ha2ked, 2023-11-03:
Sorry to say sounds like a system problem rather than the softwares problem

DerGoogler, 2023-11-04:
> Sorry to say sounds like a system problem rather than the softwares problem

@ha2ked @Pepe-454alt, that's not a system problem. The app doesn't support below Android 8 (Oreo). (`minSdk 26`)

ha2ked, 2023-11-04:
> > Sorry to say sounds like a system problem rather than the softwares problem
> 
> @ha2ked @Pepe-454alt, that's not a system problem. The app doesn't support below Android 8 (Oreo). (`minSdk 26`)

Sorry I was trying to say that but I just phrased it incorrectly

DerGoogler, 2023-11-04:
> > > Sorry to say sounds like a system problem rather than the softwares problem
> > 
> > 
> > @ha2ked @Pepe-454alt, that's not a system problem. The app doesn't support below Android 8 (Oreo). (`minSdk 26`)
> 
> Sorry I was trying to say that but I just phrased it incorrectly

@ha2ked , I opened a PR and added the requirements (#39)

ha2ked, 2023-11-04:
> @ha2ked , I opened a PR and added the requirements (#39)

Nice one thank you