simondankelmann/Bluetooth-LE-Spam #29: Not working on samsung device

Kryczyk, 2023-11-03:
Im open The apk, click start and the apk get close

Pepe-454alt, 2023-11-03:
Which version of android do you have?
I have 13 but tried on android 7 and it didn't work.

Kryczyk, 2023-11-03:
Me too have 13

Pepe-454alt, 2023-11-03:
Try to enable _My files_ and browser you are downloading from in _instal unknown applications_

Kryczyk, 2023-11-03:
Im install apk. Im enter into apk and its not working  Im click start and apk not work

Kryczyk, 2023-11-03:
Sorry for my English Im From poland

Kryczyk, 2023-11-03:
Im check this on other Phone. App work on motorola e22 but this android version is 12

Kryczyk, 2023-11-04:
V 1.0.0 not work on android 13 you can change it?

simondankelmann, 2023-11-06:
@Kryczyk 

Please try again with the latest Version, go to Settings and enable "Use Advertising with Settings". Please also try  with both devices right next to each other because the range varies from device to device. If popups appear, please close the issue :)