simondankelmann/Bluetooth-LE-Spam #31: Cant use

itsBowls, 2023-11-03:
I just clicked "start advertising" and it was immediately stopped.
[image]

ha2ked, 2023-11-03:
Has been fixed I think that was the problem but pls try the newest apk it may work

simondankelmann, 2023-11-06:
@itsBowls 

Please try again with the latest Version, go to Settings and enable "Use Advertising with Settings". Please also try  with both devices right next to each other because the range varies from device to device. If popups appear, please close the issue :)

simondankelmann, 2023-11-13:
Closed for now as the related bug was fixed in newer version

closed 2023-11-13 by simondankelmann