simondankelmann/Bluetooth-LE-Spam #37: Unable to interact with anything but the sliders

NoobToolzz, 2023-11-03:
I'm using an Android 10 on the release build of the app but can't switch the device name toggle, click the menu button on top left or click "Start advertising".

ha2ked, 2023-11-03:
Odd dont know what is wrong there

simondankelmann, 2023-11-04:
Can you try to run the app in Android Studio and share if there is anything in the log ?

GippokratOA, 2023-11-04:
Possibly because bluetooth is turned off on your device

simondankelmann, 2023-11-06:
@NoobToolzz 

Please try again with the latest Version, go to Settings and enable "Use Advertising with Settings". Please also try  with both devices right next to each other because the range varies from device to device. If popups appear, please close the issue :)

simondankelmann, 2023-11-13:
Closed for now as there is no further Information

closed 2023-11-13 by simondankelmann