simondankelmann/Bluetooth-LE-Spam #38: Apple spam

freeze1372, 2023-11-04:
Hello, very cool app! Add please Apple notifications, it is very convenient through the application. And fast, nrf is very slow.

ha2ked, 2023-11-04:
Apple may be added in the future

simondankelmann, 2023-11-04:
Please check out the latest Release :)

ntwrite, 2023-11-04:
@simondankelmann Looks great - can't get anything on Apple devices. Does it need a specific OS?

simondankelmann, 2023-11-04:
Today i saw it working on iOS 16.2 and 17.1 
Did you successfully spam any windows or android devices so we can be sure the app works properly on your phone ?

ntwrite, 2023-11-05:
@simondankelmann Tried with S21 and PXL7 but both don't really work. S21 works in sink mode partially for both Google pair and Apple, but the individual modes do not work on this device.

freeze1372, 2023-11-05:
> Сегодня я увидел, что оно работает на iOS 16.2 и 17.1. Удалось ли вам успешно спамить какие-либо устройства Windows или Android, чтобы мы могли быть уверены, что приложение работает правильно на вашем телефоне?

not really, from my point of view OnePlus 7 pro Everything is OK. but it arrives with a delay. My friend has a poko m4 pro 4g when spam starts, notifications appear on his phone 😂