simondankelmann/Bluetooth-LE-Spam #48: Spam stops when phone is turnded sideways

Schnoggi, 2023-11-04:
If you start the spam and hold the phone sideways and the display turns, the spam stops and needs to be restarted.

ha2ked, 2023-11-04:
Thanks for making this issue

Stepzor11, 2023-11-05:
@ha2ked it stops also if you lock the screen

ha2ked, 2023-11-05:
It is actually the design because much more errors could  arise if we let it run on lock screen or hold the phone sideways at that

Stepzor11, 2023-11-05:
> It is actually the design because much more errors could  arise if we let it run on lock screen or hold the phone sideways at that 

Maybe I didn't get if the first comment was sarcastic 😅, but glad that you know it 😁

ha2ked, 2023-11-05:
No you see I told him to make this on discord 👍

simondankelmann, 2023-12-30:
feature is implemented :)

closed 2023-12-30 by simondankelmann