simondankelmann/Bluetooth-LE-Spam #53: It does not work

Kalabro2311, 2023-11-04:
I have the latest version 1.0.0 that got out in 3 hours. I have poco x3 pro and it does not work on anything (windows, android, apple)
Does it need any special permissions?

simondankelmann, 2023-11-04:
Please check out this experimental version:

https://github.com/simondankelmann/Bluetooth-LE-Spam/blob/simondankelmann/experimentalAdvertising/app/debug/app-debug.apk

This needs some more testing but today i saw it working an a poco x3

simondankelmann, 2023-11-06:
@Kalabro2311 

Please try again with the latest Version, go to Settings and enable "Use Advertising with Settings". Please also try  with both devices right next to each other because the range varies from device to device. If popups appear, please close the issue :)

closed 2023-11-06 by Kalabro2311