simondankelmann/Bluetooth-LE-Spam #55: Not working

lucAmbr0, 2023-11-04:
On my Mi 11 Lite 5G (Android 13) both the app-release.apk and app-debug.apk aren't working. Tried with Samsung A54 5G and Oppo Find X3 Lite (both Android 13) and nothing appears after several attempts

simondankelmann, 2023-11-04:
Please check out this experimental version:

https://github.com/simondankelmann/Bluetooth-LE-Spam/blob/simondankelmann/experimentalAdvertising/app/debug/app-debug.apk

This needs some more testing but today i saw it working an a poco x3

lucAmbr0, 2023-11-04:
Hi, this seems to work only for a few devices on my android phones. Thank
you for your help, I'll follow your future projects :D

antonjijo, 2023-11-05:
> On my Mi 11 Lite 5G (Android 13) both the app-release.apk and app-debug.apk aren't working. Tried with Samsung A54 5G and Oppo Find X3 Lite (both Android 13) and nothing appears after several attempts

Yes I too facing this issue

ha2ked, 2023-11-05:
We are working on this issue but i would make it much easier if you could send logs pls

simondankelmann, 2023-11-06:
@antonjijo @lucAmbr0 

Please try again with the latest Version, go to Settings and enable "Use Advertising with Settings". Please also try  with both devices right next to each other because the range varies from device to device. If popups appear, please close the issue :)

antonjijo, 2023-11-06:
> @antonjijo @lucAmbr0 
> 
> Please try again with the latest Version, go to Settings and enable "Use Advertising with Settings". Please also try  with both devices right next to each other because the range varies from device to device. If popups appear, please close the issue :)

No brother I tried once more with what you said, I think you must give a update for the application, it's a request brother and thankyou for replying

lucAmbr0, 2023-11-06:
> @antonjijo @lucAmbr0 
> 
> Please try again with the latest Version, go to Settings and enable "Use Advertising with Settings". Please also try  with both devices right next to each other because the range varies from device to device. If popups appear, please close the issue :)

I tried without that setting on and it doesn't work, but when I turn it on it still doesn't work and every time a device is trying to advertise the log displays "ADVERTISE_FAILED_TOO_MANY_ADVERTISERS". this is with the release version

lucAmbr0, 2023-11-06:
> @antonjijo @lucAmbr0 
> 
> Please try again with the latest Version, go to Settings and enable "Use Advertising with Settings". Please also try  with both devices right next to each other because the range varies from device to device. If popups appear, please close the issue :)

I just tried with the debug version and again the debug version works. Also, this time the debug version works pretty well. Still not every device appears but much more and much more often than the previous version.

antonjijo, 2023-11-06:
> I just tried with the debug version and again the debug version works. Also, this time the debug version works pretty well. Still not every device appears but much more and much more often than the [...]

Okay I just try and issue if any problems appears

simondankelmann, 2023-11-06:
> I tried without that setting on and it doesn't work, but when I turn it on it still doesn't work and every time a device is trying to advertise the log displays [...]

If that happens, please just restart the app

closed 2023-11-06 by simondankelmann

antonjijo, 2023-11-06:
> If that happens, please just restart the app

Okay brother.

antonjijo, 2023-11-06:
> If that happens, please just restart the app

Brother please answer my question; From which Android version does the popups will come ?

antonjijo, 2023-11-06:
> @antonjijo @lucAmbr0 
> 
> Please try again with the latest Version, go to Settings and enable "Use Advertising with Settings". Please also try  with both devices right next to each other because the range varies from device to device. If popups appear, please close the issue :)

Brother please contact me in : antonjijo2018@outlook.com
I just sent my issue to you in a video format, contact me here at my Outlook id.

Rennanz, 2023-11-16:
The new version 1.0.4 is also not working on my phone, Android A13 android version is 13

antonjijo, 2023-11-16:
Not working (issue #55)
First at all I thank that you have mailed me . I have issued that if there is no device near me switched on blutooth it is showing that started advertising successfully . I have send you a screenshot , you can see my issue in that .

Rennanz, 2023-11-17:
Do you have an estimate of when the problem will be resolved?

Rennanz, 2023-11-17:
Log

Rennanz, 2023-11-17:
I'm in a place with no devices, and it keeps popping up. When I place a
device close by, it still doesn't work. These are the logs.

antonjijo, 2023-11-18:
WORKING BUT BUGS :[image]

 It's working now for me after I had changed some settings in victm's phone , but it's slow and showing only one or two advertisement but showing successfully advertised but only 2 or 3 only pop-uped

Rennanz, 2023-11-18:
Will you update the application on git hub?

antonjijo, 2023-11-18:
Yes I updated

Rennanz, 2023-11-18:
Can you send me the link to my email?

antonjijo, 2023-11-18:
App link ?