simondankelmann/Bluetooth-LE-Spam #63: Not Working on Xiaomi

Nothacker18282, 2023-11-05:
I have a Xiaomi 10 Lite 5G and it's not working. What can I do?

ha2ked, 2023-11-05:
## Troubleshooting Guide

Follow these steps to troubleshoot issues with the Bluetooth LE Spam app:

**Step 1: Confirm Compatibility**

- Ensure your Android device meets the requirements:
  - Android Version: 8.0 (API level 26) or higher.
  - (Verify API levels [here](<https://developer.android.com/tools/releases/platforms>)).

**Step 2: Verify App Permissions**

- Confirm that all necessary permissions (location, Bluetooth, etc.) have been granted to the app. Adjust permissions in your phone's settings; location may vary based on phone model and Android version.

**Step 3: Enable Bluetooth and Location Services**

- Turn on Bluetooth and Location Services in your device's settings.

**Step 4: Check for Latest Release**

- Ensure you are using the latest release. Find it in the [Releases Section](<https://github.com/simondankelmann/Bluetooth-LE-Spam/releases>).

**Still experiencing issues?**

- If issues persist, gather logs using Android Studio's LogCat. Refer to this [Video Tutorial](<https://www.youtube.com/watch?v=smOzLSTaso0>) for guidance.

Please provide us with the following information via direct message:

- Phone Model:
- Android Version:
- Detailed Issue Description:
- Screenshots/Videos (if possible):
- Log Files:

We greatly appreciate your cooperation in troubleshooting. Gathering logs may seem a bit complicated, but it plays a crucial role in improving the app for everyone's benefit. Thank you for your contribution to the community!

**Important Note**: Please be aware that the effectiveness of the Bluetooth LE Spam functionality may vary depending on factors such as your specific Android device, as well as the device receiving the advertisements. This project is primarily a proof of concept and results may differ in different scenarios. We appreciate your understanding and welcome any feedback or insights you may have!

smdkx, 2023-11-08:
Redmi Note 5 Android 13 also doesn't work. With "Use Advertising with Settings" enabled, it displays an error

wh1ter0z, 2023-11-11:
Its for google phones not xiaomi i think so

simondankelmann, 2023-11-13:
Closed for now as there is no further Information