simondankelmann/Bluetooth-LE-Spam #73: it doesn't work that far

selenium2106, 2023-11-05:
The effectiveness of spam is not so much at a greater distance, being less than 10cm away it resulted in many advertisements, but more than that, the messages are not as frequent or practically non-existent. Sorry if there are spelling errors, I use a translator

ha2ked, 2023-11-05:
if you got a external bluetooth adapter it would be better

selenium2106, 2023-11-05:
You just have to connect it through OTG and that's it? Or do you have to do additional configuration?

Justxd22, 2023-11-06:
Needs kernel support for loading modules/usb support

simondankelmann, 2023-11-06:
Thanks for sharing the Information, currently i think there is no way to extend the range using the android sdk. therefore i will close this issue for now.

closed 2023-11-06 by simondankelmann