simondankelmann/Bluetooth-LE-Spam #78: Add ability to change tx power but on rooted devices

chronosirius, 2023-11-07:
I just wanted to know if this is possible - if it is I'll look into it and maybe (?) make a PR.

simondankelmann, 2023-11-07:
Feel free to look into it, until now I didn't find a way to do that even with root unless you have direct access to the hci device or BusyBox Installed. But this app should be a standalone application :)

chronosirius, 2023-11-07:
If one has root access to the shell, there is a way to get access to the hci device, right?

simondankelmann, 2023-11-08:
Not necessarily, i have looked at it a few minutes on an older rooted phone but didnt find anything. I think to address a bigger number of devices a user would need root access and hcitool installed which comes with busybox in some cases i think ?! if thats the case i think it would be possible to send actual bytedata directly via command line allowing us to modifiy the payload completely free :)

chronosirius, 2023-11-08:
Could one clone the hcitool binary and package it with the app or, while keeping it a standalone app, tell users that one can do more with busybox if they *choose* to install it?

DerGoogler, 2023-11-12:
> Not necessarily, i have looked at it a few minutes on an older rooted phone but didnt find anything. I think to address a bigger number of devices a user would need root access and hcitool installed [...]

Magisk and KernelSU comes with BusyBox installed. You also can it build it yourself and include it

chronosirius, 2023-11-13:
Interesting, I don't have hcitool or bluez installed on my magisk phone.

DerGoogler, 2023-11-17:
@simondankelmann take a look at this thread: https://stackoverflow.com/a/35499446.

simondankelmann, 2023-11-17:
Very Interesting ! But i think this way you can only set the actual txpower level. What we need to increase the range is setting the tx power in the payload to something like "low" but actually send with "high"

Fusseldieb, 2024-01-21:
I'm looking forward to this, too. 
I currently have 2 rooted devices (Magisk), a S7E and a S10e. 
Would love to see this implemented.

TailsFanLOL, 2024-03-08:
Is this possible with Shizuku?

chronosirius, 2024-03-11:
I think the problem with this is that Android, since version 4.4 has been using BlueDroid as a Bluetooth stack. BlueDroid handles the tx bit by itself, so we cannot control it without replacing the Bluetooth stack completely. I was thinking perhaps we could bundle BlueZ with the app and then compile it on first run, but I'm not sure exactly how we would hook BlueZ in.

TLDR: If we want this we will have to use a different Bluetooth stack than the android native one. Root is unhelpful without that.

DerGoogler, 2024-03-11:
> I think the problem with this is that Android, since version 4.4 has been using BlueDroid as a Bluetooth stack. BlueDroid handles the tx bit by itself, so we cannot control it without replacing the [...]

Just fork that part

Alluseri, 2024-03-18:
I presume it would not be possible to use e.g. BlueZ on newer versions of Android? Maybe through some kind of Magisk patch(that doesn't exist to my knowledge)?

chronosirius, 2024-05-13:
> I presume it would not be possible to use e.g. BlueZ on newer versions of Android? Maybe through some kind of Magisk patch(that doesn't exist to my knowledge)?

I know the BlueZ maintainers had an android version up to version 5.0, but after that I have found nothing, not even magisk patches. One would have to completely modify AOSP to change the stack, and I have no idea how to even begin doing that.

ha2ked, 2024-05-15:
None of the original devs are up for doing this so if you want to do it yourself, go ahead.

closed 2024-11-16 by ha2ked