simondankelmann/Bluetooth-LE-Spam #89: App freezes

Nabeel-javaid, 2023-11-08:
I installed both of the versions and after trying it for 1 time when i tried to use it again the UI freeze and i wasn't able to do anything.. Tried it like 10 times and same thing happened.

simondankelmann, 2023-11-08:
Did you also try activating the Switch in the Settings ?

Nabeel-javaid, 2023-11-08:
> Did you also try activating the Switch in the Settings ?

As soon as I open the app. It freezes, I'm not even able to enable that switch in the setting

simondankelmann, 2023-11-08:
Ok i understand, is there any chance you can run it in android studio and see the logs ?
what phone and android version do you use ?
maybe you can also try granting all permission manually in the app settings.

Nabeel-javaid, 2023-11-08:
I'm using infinix mobile phone and android version 10.

Currently I'm not able to use Android studio but maybe in future i can try and yes I've given all the permissions to the app

simondankelmann, 2023-11-27:
Please try again with the latest Release

closed 2023-11-27 by simondankelmann