simondankelmann/Bluetooth-LE-Spam #94: Iphone not working on Samsung galaxy note9

kingexplsion7, 2023-11-08:
I have a samsung galaxy note 9 and i use the apple bluetooth popup and it doesn't work. The android works but the apple doesn't please help me on why and if it has to do with my bluetooth drive.

simondankelmann, 2023-11-08:
Please try again after enabling the Switch in the Settings

kingexplsion7, 2023-11-09:
It now shows apple tv popup but the airpods or headphone doesn't popped up. It only popped up once. Its still not working. Or it might just be the iphone is blocking the spamming.

ha2ked, 2023-11-11:
Its very hit and miss

ha2ked, 2023-11-11:
> It now shows apple tv popup but the airpods or headphone doesn't popped up. It only popped up once. Its still not working. Or it might just be the iphone is blocking the spamming.

I believe that to be true

simondankelmann, 2023-11-13:
Closed for now, iOs seems to be blocking Popups after a certain amount of Advertisers

See: #Conclusion

https://www.mobile-hacker.com/2023/11/08/android-kitchen-sink-send-ble-spam-to-ios-android-and-windows-at-once-using-android-app/

closed 2023-11-13 by simondankelmann