simondankelmann/Bluetooth-LE-Spam #106: Not working

antonjijo, 2023-11-13:
I have tried from 1st version to upto the latest version application but if i click start it's showing successfully advertised but there is no popups after I switched on blutooth also and also showing successfully advertised when there is no device which has blutooth switched on , why it is showing fake successful advertised ? Reply me....

simondankelmann, 2023-11-13:
What did you try to Spam, iOs (Modals or Devices), Android, or Windows ? Please try with the sending Device next to the receiver. For the sender it is not important if the receiver has bluetooth enabled. once the data is sent out it is marked as success. there is no check if any device did receive that.

antonjijo, 2023-11-16:
> What did you try to Spam, iOs (Modals or Devices), Android, or Windows ? Please try with the sending Device next to the receiver. For the sender it is not important if the receiver has bluetooth [...]

I have tried to advertise Android devices.As you have telling me to try with my reciveing device near to sender device but I have already tried it.

qasimkhn, 2023-11-17:
https://youtu.be/KDqY7yST7Kw?feature=shared,,,check it out brother,,just enable few settings then try ,,it will 100% work

simondankelmann, 2023-12-30:
closed for now as there are no further information

closed 2023-12-30 by simondankelmann