simondankelmann/Bluetooth-LE-Spam #119: It Doesnt work i even tried it on my friends android and anothers iphone it didnt work

HasanRav, 2023-11-17:


qasimkhn, 2023-11-18:
https://youtu.be/KDqY7yST7Kw?feature=shared just try to check the settings

l3d00m, 2023-11-21:
> https://youtu.be/KDqY7yST7Kw?feature=shared just try to check the settings 

Stop spamming your YouTube Video under every issue. @simondankelmann can you just ban them?

qasimkhn, 2023-11-21:
> > https://youtu.be/KDqY7yST7Kw?feature=shared just try to check the settings
> 
> Stop spamming your YouTube Video under every issue. @simondankelmann can you just ban them?

its not spammming brother,,,,i am just clearing to the guys that its working fine without any issue,,some people dont know which settings need to be enabled to let it work that what that video explain,,,,hope u get the answer and hope you will kepp ur mouth shut and when u will open it nice words gona come out from there ,,,have a gr8 day ahead

tiagorangel2011, 2024-02-01:
Try deleting your cache and app data (https://github.com/simondankelmann/Bluetooth-LE-Spam/issues/206#issuecomment-1876900828), it should work if you have default settings.

closed 2024-10-25 by ha2ked