simondankelmann/Bluetooth-LE-Spam #120: too many advertisers error

HikaruIchijyo, 2023-11-19:
[image]

Idk whether it is error or something. Can anybody tell me how to make it work

ha2ked, 2023-11-19:
Will get back to you on it

ha2ked, 2023-11-19:
Turn on legacy advertising slow down the advertising and make sure Bluetooth is on

HikaruIchijyo, 2023-11-19:
Already on
Tried with max delay.
Bluetooth already on

ha2ked, 2023-11-19:
legacy advertising?

ha2ked, 2023-11-19:
is it on that is

HikaruIchijyo, 2023-11-19:
> is it on that is

Yep

ha2ked, 2023-11-19:
Odd

simondankelmann, 2023-11-19:
Please try again with the latest Release:

https://github.com/simondankelmann/Bluetooth-LE-Spam/releases

HikaruIchijyo, 2023-11-19:
> Please try again with the latest Release:
> 
> https://github.com/simondankelmann/Bluetooth-LE-Spam/releases

Still same

[image]

simondankelmann, 2023-11-19:
whats your interval and avertising mode now ?

HikaruIchijyo, 2023-11-19:
> whats your interval and avertising mode now ?

10000 ms & kitchen sync.
I tried all of them

simondankelmann, 2023-11-19:
do you have legacy mode enabled ?

also, are you connected with any other bluetooth devices ?

HikaruIchijyo, 2023-11-19:
> do you have legacy mode enabled ?
> 
> also, are you connected with any other bluetooth devices ?

Yes.

No

HikaruIchijyo, 2023-11-19:
Possible to get logs or something fr u?

simondankelmann, 2023-11-19:
if you could provide those logs i can look if there is something odd. but this issue usually appears if too many processes try to advertise at once. if you turn off bluetooth and turn it back on, and then start advertising with a delay of 1000, do you immidiately get the error at the first try or does it let you send a few times and then stops ?

HikaruIchijyo, 2023-11-20:
> if you could provide those logs i can look if there is something odd. but this issue usually appears if too many processes try to advertise at once. if you turn off bluetooth and turn it back on, [...]

Immediately

HikaruIchijyo, 2023-11-20:
https://bin.kv2.dev/~655ae7fa5de7ed6bfdba5c45

simondankelmann, 2023-11-20:
do you have bluetooth turned on ? thats weird...

HikaruIchijyo, 2023-11-20:
> do you have bluetooth turned on ? thats weird... 

Yep it is

HikaruIchijyo, 2023-11-20:
> do you have bluetooth turned on ? thats weird... 

W8 sorry but my watch is connected to phone.
Is it causing the issue.

I just remembered that it is also connected🥹

HikaruIchijyo, 2023-11-20:
Still same.
I just unpaired it but same

HikaruIchijyo, 2023-11-21:
Still same
No watch
No other device
Still that error

simondankelmann, 2023-11-27:
From the Screenshot you provided it seems your device is connected with some bluetooth device. other than that i dont know what could cause the problem. maybe you have nrf connect running or something ?  i guess you have already rebooted the phone?

very strange because the app only needs one advertisement channel

simondankelmann, 2023-11-27:
what exact phone do you use ?

HikaruIchijyo, 2023-11-28:
> From the Screenshot you provided it seems your device is connected with some bluetooth device. other than that i dont know what could cause the problem. maybe you have nrf connect running or [...]

Yep tried multiple times

HikaruIchijyo, 2023-11-28:
> what exact phone do you use ?

Redmi note 10s on custom android 13.

Is there anyway to get logs?
I am rooted though so can get logs but last time i checked there was nothing

HikaruIchijyo, 2023-11-28:
Somehow it got fixed automatically.

I am on debug rn if it happens again i wkll just reopen it

closed 2023-11-28 by HikaruIchijyo