simondankelmann/Bluetooth-LE-Spam #124: ADVERTISE_FALED_DATA_TOO_LARGE

BiskottoScriptato, 2023-11-20:
Getting this error when spamming apple devices, legacy switch on, 1000ms

simondankelmann, 2023-11-27:
Does your device support Bluetooth 5 ? If yes, please try with switching Legacy to off.

BiskottoScriptato, 2023-11-27:
> Does your device support Bluetooth 5 ? If yes, please try with switching Legacy to off.

same thing on off.

BiskottoScriptato, 2023-11-27:
> which app version do you use?

the latest one 1.0.5

shakir2117, 2023-11-30:
ADVERTISE_FALED_DATA_TOO_LARGE i am also getting this issue i tried turning of the legacy but then also i am getting this error and i am using letest version of the app and bluetooth version is 5

000exploit, 2024-01-31:
Nothing on this? The same on OnePlus 5T with LineageOS 20.

simondankelmann, 2024-02-01:
Sry, since i wasnt able to reproduce this issue and due to less time i wasnt able to fix it yet.

ha2ked, 2024-02-10:
> Getting this error when spamming apple devices, legacy switch on, 1000

What device are you on and do you have any ROMs installed

closed 2024-10-25 by ha2ked