simondankelmann/Bluetooth-LE-Spam #140: ADVERTISE_FAILED_TOO_MANY_ADVERTISERS getting this error

shakir2117, 2023-11-30:


simondankelmann, 2023-11-30:
Please close the app, turn off bluetooth and turn it back on. then try again

shakir2117, 2023-11-30:
i did and it started working but it is still giving error for IOS device popups list but and modal list ,rather than that every thing is working

NjuuOwO, 2023-12-02:
Same issue, using latest release (1.0.5), phone Redmi Note 9 on LineageOS 13 latest build.
Restarting phone allows me to run it once, but after refreshing/restarting the spam it gives error above.

NjuuOwO, 2023-12-03:
> can someone one you test that on your devices somehow ?

How can I do it?

SashaLore, 2024-03-29:
I was able to partially fix this. After a lot of spam with packets, when this error appears, I rebooted the phone with bluetooth turned on and immediately start BLE spam again after reboot. The downside is that spam works only once, and without rebooting the device it’s impossible to run it a second time in a row

bzbrn, 2024-08-14:
I also use Redmi Note 9 with LineageOS, and I get this error too. The reboot method works for me as well, but as noted above, it only works once.

exrazer, 2024-11-11:
Hi me too I have this error no matter what I do if you want I can send you a video of my screen.

ha2ked, 2024-11-14:
This error is most likely caused by a custom ROM and has not fix