simondankelmann/Bluetooth-LE-Spam #145: app_debug VS app_release

GeorgeYT9769, 2023-12-02:
What is better to get? app_debug or app_release? Tried app_release and didn't work.

ha2ked, 2023-12-02:
What do you mean by "didn't work"

GeorgeYT9769, 2023-12-02:
> What do you mean by "didn't work"  

I installed app_release and it didn't work. So I am wondering if app_debug works and if it works/doesn't work for other people.

ha2ked, 2023-12-03:
> > What do you mean by "didn't work"
> 
> I installed app_release and it didn't work. So I am wondering if app_debug works and if it works/doesn't work for other people.

There is little to no difference between them so if one didn't work then the other one wont either

GeorgeYT9769, 2023-12-06:
> can you be more precise on what exactly did not work ? did the app not open at all, did the app work but you had no popups on target devices etc...

Popups didn't show, and I don't think it's my device's issue. nRF app where I had the setup for advertising worked. But also many people show that your app worked. Might be an issue between app and Bluetooth adapter...

closed 2023-12-16 by GeorgeYT9769