simondankelmann/Bluetooth-LE-Spam #146: my own pair connections

gpoveda, 2023-12-03:
Hi all,

Im sorry, I would like to know if somebody can just advise me how can I create my pair device in the collections. I mean,  I would like to create an item on the list, connect  to it and open the modal for going to the google store for my own app.

I can see there is a code like         "E57B57" to "Super Device", which is decoded and store to a sqlite db.

By the way, when I try to look for the db file on the android device I can not find it, how can I search it?

I really appreciate all your help. thank you very much

simondankelmann, 2023-12-03:
Maybe you can refer to this :

https://www.youtube.com/watch?v=mmTvqyIoxVo

the database file is stored somewhere in the apps data space, afaik thats not easily accessable within android

closed 2023-12-03 by simondankelmann

gpoveda, 2023-12-06:
Hi, 

Thank you very much for you quick response and your advise. Sorry. one question more please: How can I add my own id device for the Bluetooth-LE-Spam (android app)?. 

The video is very clear, but, they  use a physical device (firmware).