simondankelmann/Bluetooth-LE-Spam #167: Enhance Device List Organization by Brand Categorization and ASCII Sorting

lc6464, 2023-12-13:
I've been exploring the Bluetooth-LE-Spam app and appreciate the effort put into its development. During use, I've noticed that the device list within the app appears to be somewhat disorganized, which makes the user experience less efficient than it could be.

To improve the usability and efficiency of the device search function, I would like to suggest a couple of enhancements:

- **Brand Categorization**: It would be immensely helpful if the devices could be categorized by their respective brands. This way, users can quickly navigate to a specific brand and find the devices associated with it.
- **ASCII Sorting**: Once categorized, sorting the devices within each brand category in ASCII order would further streamline the searching process. This provides a predictable and alphabetical ordering method, allowing users to effortlessly scroll to the desired device.

Implementing these changes could significantly enhance the user experience by making it easier and faster to find specific devices. I understand that these modifications may require some additional work, but I believe that the benefits in terms of app usability would be substantial.

Thank you for considering these suggestions. I look forward to any thoughts or feedback you might have on this proposal.

simondankelmann, 2023-12-14:
Thats a good Idea :)

simondankelmann, 2023-12-14:
Merged your PR with sorted FastPairing Devices, thanks for your contribution !

closed 2025-01-23 by ha2ked