simondankelmann/Bluetooth-LE-Spam #175: Add support without needing to enable location for victim devices

Darkrevengehack, 2023-12-15:
They need to add what I told you in the title to the app, and also add compatibility against those devices that do not have or require the Google neaby service for the app to work bluetooth spam.

simondankelmann, 2023-12-16:
What exactly do you mean ?

Darkrevengehack, 2023-12-17:
> Qué quieres decir exactamente ?

Hablo más que nada sobre el neaby share de google que se necesita habilitar para que funcione el bluetooth le spam en el dispositivo, y lo mismo con la ubicación, la ubicación hay que activarla en tu dispositivo y en el dispositivo víctima para que funcione el spam, y no es correcto, debería de funcionar con el bluetooth y la ubicación de tu dispositivo no del dispositivo víctima y sin necesidad del google neaby share.

simondankelmann, 2023-12-30:
closed, because its not possible to trigger google nearby notifications without the receving side having the feature turned on

closed 2023-12-30 by simondankelmann