simondankelmann/Bluetooth-LE-Spam #176: App close immediately after start

sztuk5, 2023-12-15:
App is instantly closing when trying to open after update (1.0.7 version with data previously cleared) on Samsung S22 Ultra.

simondankelmann, 2023-12-16:
Can you provide some logs ?

sztuk5, 2023-12-16:
> Can you provide some logs ?

How can I do that?

simondankelmann, 2023-12-16:
Best way would be:

clone the repo
open in android studio
connect your phone via adb
run the app through android studio on your phone
check logcat monitor for error messages

OlekAleksander, 2023-12-16:
> App is instantly closing when trying to open after update (1.0.7 version with data previously cleared) on Samsung S22 Ultra.

I had the same issue. Uninstalling and reinstalling the app worked for me.

ha2ked, 2023-12-16:
You could also clear app data and that will fix it

sztuk5, 2023-12-16:
@OlekAleksander @ha2ked 
I tried multiple times to clear data and reinstall or just reinstall and it didn't work. Now I tried to clear data, uninstall, install again and clear data and it worked. Thank You!

closed 2023-12-16 by simondankelmann