simondankelmann/Bluetooth-LE-Spam #187: iOS Action Modals and iOS 17 Crash lists have exactly the same actions

nmac101, 2023-12-16:
I don't think it's supposed to. I attached a long screenshot.

[image]

ha2ked, 2023-12-16:
See it's just a change of interval if you want to crash

simondankelmann, 2023-12-16:
actually even the payload is modified. it is intended to use the same actions as in the other list but the payload is appended with 0x000010{3 bytes random}.

closed 2023-12-16 by simondankelmann

nmac101, 2023-12-16:
> actually even the payload is modified. it is intended to use the same actions as in the other list but the payload is appended with 0x000010{3 bytes random}.

wait, didnt apple fix this? https://www.zdnet.com/article/ios-17-2-update-puts-an-end-to-flipper-zero-iphone-shenanigans/
i dont have an iphone to test this btw