simondankelmann/Bluetooth-LE-Spam #188: 1.0.7 crashes

GeorgeYT9769, 2023-12-16:
After latest update 1.0.7 the app asks for permission to use Bluetooth and crashes.

Altuskha333, 2023-12-16:
Clear all data in applications if you have MIUI

simondankelmann, 2023-12-17:
Please delete your app data before using the update

GeorgeYT9769, 2023-12-17:
> Please delete your app data before using the update

I did, not even reinstalling the app works.

simondankelmann, 2023-12-17:
Then please post some logs, otherwise I won't be able to work on this

GeorgeYT9769, 2023-12-17:
Here, a video of the crash: https://youtube.com/shorts/IOWd6QofwqQ?si=a3zGbqScpMCAvA_j

leafymar, 2023-12-18:
it crashes on start:

java.lang.IllegalArgumentException: Can't convert value to enum, unknown value: ADVERTISEMENT_TYPE_CONTINUITY_DEVICE_POPUPS
	at de.simon.dankelmann.bluetoothlespam.Database.Dao.AdvertisementSetDao_Impl.__AdvertisementSetType_stringToEnum(AdvertisementSetDao_Impl.java:603)
	at de.simon.dankelmann.bluetoothlespam.Database.Dao.AdvertisementSetDao_Impl.getAll(AdvertisementSetDao_Impl.java:243)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment$checkDatabase$1.invokeSuspend(StartFragment.kt:517)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at kotlinx.coroutines.internal.LimitedDispatcher.run(LimitedDispatcher.kt:42)
	at kotlinx.coroutines.scheduling.TaskImpl.run(Tasks.kt:95)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:570)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:677)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:664)
	Suppressed: kotlinx.coroutines.DiagnosticCoroutineContextException: [StandaloneCoroutine{Cancelling}@8deccfb, Dispatchers.IO]

simondankelmann, 2023-12-18:
Those logs show that the old database is not properly removed. Not all App data is deleted.

leafymar, 2023-12-18:
> Those logs show that the old database is not properly removed. Not all App data is deleted.

how do i delete all of the app data?
I've already deleted the data from settings and it still crashes?

GeorgeYT9769, 2023-12-18:
> Those logs show that the old database is not properly removed. Not all App data is deleted.

How to delete all data? I deleted the data I could, reinstalled it, nothing... Also how to get the log?

StaiMeer, 2023-12-18:
Just go to file manager, Android, data , de.xxxxx.bluetoohtlespamxxxxxxx (idk the name, something with Blspam); Delete this entire folder and you're good to go

leafymar, 2023-12-18:
> > > > > Those logs show that the old database is not properly removed. Not all App data is deleted.
> > > > 
> > > > 
> > > > how do i delete all of the app data?
> > > > I've already deleted the data from settings and it still crashes?
> > > 
> > > 
> > > Try to quit the app, and activate the GPS before enter again, be sure to give all required permissions to the app.
> > 
> > 
> > Its still crashing i dont think it can be fixed??
> 
> Idk, I installed the app but it crashed, so I uninstalled it and reinstalled it, but the problem remained, so I went into the app settings and deleted the data, then I manually set the permissions in the settings and also activated the GPS in the cell phone toggles together with the Bluetooth, when I reopened the app it worked again, and it didn't give me any more problems.

Thank you!! It works now!! I set the permissions manually and it worked! Thank you so much 🙏🙏

GeorgeYT9769, 2023-12-19:
> Idk, I installed the app but it crashed, so I uninstalled it and reinstalled it, but the problem remained, so I went into the app settings and deleted the data, then I manually set the permissions [...]

On my side the problem remains, anything I do doesn't help.

GeorgeYT9769, 2023-12-19:
After doing the same things written here again and again, it started working. LOL

closed 2023-12-19 by GeorgeYT9769

simondankelmann, 2023-12-19:
Good to know it is finally working