simondankelmann/Bluetooth-LE-Spam #201: WearOS make app dpi smaller

srt54558, 2023-12-20:
Hello dear BLE spam Community/developers/users!
I've sideloaded BLE spam on my Pixel watch 2 (wifi) and it worked. But the UI is horrible. Can you please just make it a bit smaller on small screens? Like changing the dpi?

[image]
[image]

I can indeed enter the settings and change them. And I can scroll the home screen, but when I select a category, I can't look at the folders, devices to emulate or even the randomize button, etc. I can't scroll on this screen. Only select what possible to see. (Pic 2)

Would be nice to make it fit. Maybe even dark mode for the watch (AMOLED black if possible). I don't know if it's running in the background at the moment

Great app, even crashed iOS 17! Thanks alot man!

P.s.: on android the monochrome icon is missing

srt54558, 2023-12-20:
Okay, no it doesn't work for some reason. I can start it and tried in in legacy and not legacy but it doesn't show up on Windows

GammaPotato, 2023-12-21:
Nice idea. I will sidelad the app too and test it on IOS, Android and windows. Im try it with galaxy watch 5. Whats ur watch?

ha2ked, 2025-01-23:
I can't imagine this would work without a rewrite and that is not an option but anyone is welcome to do so.