simondankelmann/Bluetooth-LE-Spam #204: Application can not launch

lc6464, 2023-12-21:
Version: 1.0.7
Device: Xiaomi 13 (2211133C)
System: MIUI 14.0.5.0 (UMCCNXM)
```
java.lang.IllegalArgumentException: Can't convert value to enum, unknown value: ADVERTISEMENT_TYPE_CONTINUITY_DEVICE_POPUPS
	at de.simon.dankelmann.bluetoothlespam.Database.Dao.AdvertisementSetDao_Impl.__AdvertisementSetType_stringToEnum(AdvertisementSetDao_Impl.java:603)
	at de.simon.dankelmann.bluetoothlespam.Database.Dao.AdvertisementSetDao_Impl.getAll(AdvertisementSetDao_Impl.java:243)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment$checkDatabase$1.invokeSuspend(StartFragment.kt:517)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at kotlinx.coroutines.internal.LimitedDispatcher.run(LimitedDispatcher.kt:42)
	at kotlinx.coroutines.scheduling.TaskImpl.run(Tasks.kt:95)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:570)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:677)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:664)
	Suppressed: kotlinx.coroutines.DiagnosticCoroutineContextException: [StandaloneCoroutine{Cancelling}@303d59a, Dispatchers.IO]
```

simondankelmann, 2023-12-21:
Please delete all App data and cache

MustafaErenKaplan, 2023-12-21:
Use debug version

lc6464, 2023-12-21:
> Please delete all App data and cache 

Thank you. It is working now.

lc6464, 2023-12-21:
> I have same issue on Galaxy Z fold 3. Android 13. App crashes instant. Android system tells me that the app have an issue and cant start

You can try to clear all data of the application.

closed 2023-12-21 by lc6464