simondankelmann/Bluetooth-LE-Spam #206: Not working

Ismail20081206, 2023-12-25:
Using S22 ultra

slashy, 2023-12-27:
Can confirm. It does nothing when I start it on my S22 Ultra. It works from my flipper tho, so the phones I tried it on are definitely vulnerable to it.

Ismail20081206, 2023-12-27:
It used to work on my S22 Ultra but doesn't since the last 2 updates

ha2ked, 2024-01-03:
Okay............. really odd will look into it

simondankelmann, 2024-01-03:
Can you please tell what exact steps you did to try it ?

slashy, 2024-01-03:
> Can you please tell what exact steps you did to try it ?

Well, uhm, sure. But it's not like there's actually much to setup to begin with.

- Opened the App. 
- Selected the iOS Playlist (Continuity)
- Pressed play
- ... 
- Nothing happend in an iPhone (iOS 17.x (not 17.2 but even then there should be at least some popups))
- Selected Samsung Playlist (Easy Setup)
- Pressed play
- ....
- Nothing happend on a Samsung Galaxy S23 Ultra nor Samsung Galaxy S7+ nor Samsung Fold
- Selected Android Playlist (Fast Pair)
- Pressed play
- ....
- Nothing happend on a Samsung Galaxy S23 Ultra nor Samsung Galaxy S7+ nor Samsung Fold 
- Tried the above again with Legacy Advertising
- ....
- Nothing happend on any device

All with TX power set to max. Flipper worked on the same devices and produced popups.

slashy, 2024-01-03:
I also remember that it used to work on my Galaxy S7+ when I first tried it.

slashy, 2024-01-03:
> I also remember that it used to work on my Galaxy S7+ when I first tried it.
> 
> Edit: I just tried an older version (v1.0.5) and it works at least on the Galaxy S23 Ultra with both the Android and Samsung Playlist

Okay, now this is where it gets really weird. I just reinstalled the latest version and it seems to work again with both playlists on the same phone (S23 Ultra). I can't test iPhone though as I don't have a device near me at the moment.

Not sure what's causing the issue, but I tried it last week and it certainly did not work on the very same device.

matosdiego, 2024-01-04:
Delete the cache and all the application data in the case of Android and the APP works again

tiagorangel2011, 2024-02-01:
> Delete the cache and all the application data in the case of Android and the APP works again

Can confirm this solution works.

Zedasix, 2024-03-14:
Same on Pixel 7a

tiagorangel2011, 2024-03-16:
Clear your cache and app data. Works most of the time

ha2ked, 2024-04-01:
If that doesn't work try the debug version