simondankelmann/Bluetooth-LE-Spam #214: Crash upon start

Doubble29x, 2024-01-05:
`
type: crash
osVersion: google/raven/raven:14/UQ1A.240105.002/2024010400:user/release-keys
package: de.simon.dankelmann.bluetoothlespam:1
process: de.simon.dankelmann.bluetoothlespam
processUptime: 172 + 284 ms
installer: com.android.packageinstaller

java.lang.RuntimeException: Unable to start activity ComponentInfo{de.simon.dankelmann.bluetoothlespam/de.simon.dankelmann.bluetoothlespam.MainActivity}: android.view.InflateException: Binary XML file line #15 in de.simon.dankelmann.bluetoothlespam:layout/activity_main: Binary XML file line #19 in de.simon.dankelmann.bluetoothlespam:layout/content_main: Error inflating class fragment
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3833)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3974)
	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:103)
	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:139)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:96)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2475)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loopOnce(Looper.java:205)
	at android.os.Looper.loop(Looper.java:294)
	at android.app.ActivityThread.main(ActivityThread.java:8279)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
	at com.android.internal.os.ExecInit.main(ExecInit.java:49)
	at com.android.internal.os.RuntimeInit.nativeFinishInit(Native Method)
	at com.android.internal.os.RuntimeInit.main(RuntimeInit.java:359)
Caused by: android.view.InflateException: Binary XML file line #15 in de.simon.dankelmann.bluetoothlespam:layout/activity_main: Binary XML file line #19 in de.simon.dankelmann.bluetoothlespam:layout/content_main: Error inflating class fragment
Caused by: android.view.InflateException: Binary XML file line #19 in de.simon.dankelmann.bluetoothlespam:layout/content_main: Error inflating class fragment
Caused by: java.lang.SecurityException: Permission Denial: starting Intent { act=android.bluetooth.adapter.action.REQUEST_ENABLE cmp=com.android.settings/.bluetooth.RequestPermissionActivity } from ProcessRecord{f9ec0a9 27854:de.simon.dankelmann.bluetoothlespam/u0a190} (pid=27854, uid=10190) requires android.permission.BLUETOOTH_CONNECT
	at android.os.Parcel.createExceptionOrNull(Parcel.java:3066)
	at android.os.Parcel.createException(Parcel.java:3050)
	at android.os.Parcel.readException(Parcel.java:3026)
	at android.os.Parcel.readException(Parcel.java:2968)
	at android.app.IActivityTaskManager$Stub$Proxy.startActivity(IActivityTaskManager.java:2086)
	at android.app.Instrumentation.execStartActivity(Instrumentation.java:1879)
	at android.app.Activity.startActivityForResult(Activity.java:5661)
	at androidx.activity.ComponentActivity.startActivityForResult(ComponentActivity.java:780)
	at androidx.core.app.ActivityCompat$Api16Impl.startActivityForResult(ActivityCompat.java:809)
	at androidx.core.app.ActivityCompat.startActivityForResult(ActivityCompat.java:246)
	at androidx.activity.ComponentActivity$1.onLaunch(ComponentActivity.java:239)
	at androidx.activity.result.ActivityResultRegistry$2.launch(ActivityResultRegistry.java:173)
	at androidx.fragment.app.Fragment$10.launch(Fragment.java:3632)
	at androidx.activity.result.ActivityResultLauncher.launch(ActivityResultLauncher.java:47)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment.promptEnableBluetooth(StartFragment.kt:565)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment.checkBluetoothAdapter(StartFragment.kt:553)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment.onCreateView(StartFragment.kt:86)
	at androidx.fragment.app.Fragment.performCreateView(Fragment.java:3114)
	at androidx.fragment.app.FragmentStateManager.createView(FragmentStateManager.java:557)
	at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:272)
	at androidx.fragment.app.FragmentStore.moveToExpectedState(FragmentStore.java:114)
	at androidx.fragment.app.FragmentManager.moveToState(FragmentManager.java:1455)
	at androidx.fragment.app.FragmentManager.dispatchStateChange(FragmentManager.java:3034)
	at androidx.fragment.app.FragmentManager.dispatchViewCreated(FragmentManager.java:2945)
	at androidx.fragment.app.Fragment.performViewCreated(Fragment.java:3148)
	at androidx.fragment.app.FragmentStateManager.ensureInflatedView(FragmentStateManager.java:410)
	at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:271)
	at androidx.fragment.app.FragmentLayoutInflaterFactory.onCreateView(FragmentLayoutInflaterFactory.java:142)
	at androidx.fragment.app.FragmentController.onCreateView(FragmentController.java:136)
	at androidx.fragment.app.FragmentActivity.dispatchFragmentsOnCreateView(FragmentActivity.java:247)
	at androidx.fragment.app.FragmentActivity.onCreateView(FragmentActivity.java:226)
	at android.view.LayoutInflater.tryCreateView(LayoutInflater.java:1081)
	at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:1009)
	at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:973)
	at android.view.LayoutInflater.rInflate(LayoutInflater.java:1135)
	at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:1096)
	at android.view.LayoutInflater.parseInclude(LayoutInflater.java:1275)
	at android.view.LayoutInflater.rInflate(LayoutInflater.java:1131)
	at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:1096)
	at android.view.LayoutInflater.parseInclude(LayoutInflater.java:1275)
	at android.view.LayoutInflater.rInflate(LayoutInflater.java:1131)
	at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:1096)
	at android.view.LayoutInflater.inflate(LayoutInflater.java:694)
	at android.view.LayoutInflater.inflate(LayoutInflater.java:538)
	at de.simon.dankelmann.bluetoothlespam.databinding.ActivityMainBinding.inflate(ActivityMainBinding.java:53)
	at de.simon.dankelmann.bluetoothlespam.databinding.ActivityMainBinding.inflate(ActivityMainBinding.java:47)
	at de.simon.dankelmann.bluetoothlespam.MainActivity.onCreate(MainActivity.kt:62)
	at android.app.Activity.performCreate(Activity.java:8674)
	at android.app.Activity.performCreate(Activity.java:8652)
	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1461)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3815)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3974)
	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:103)
	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:139)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:96)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2475)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loopOnce(Looper.java:205)
	at android.os.Looper.loop(Looper.java:294)
	at android.app.ActivityThread.main(ActivityThread.java:8279)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
	at com.android.internal.os.ExecInit.main(ExecInit.java:49)
	at com.android.internal.os.RuntimeInit.nativeFinishInit(Native Method)
	at com.android.internal.os.RuntimeInit.main(RuntimeInit.java:359)
Caused by: android.os.RemoteException: Remote stack trace:
	at com.android.server.wm.ActivityTaskSupervisor.checkStartAnyActivityPermission(ActivityTaskSupervisor.java:1136)
	at com.android.server.wm.ActivityStarter.executeRequest(ActivityStarter.java:1081)
	at com.android.server.wm.ActivityStarter.execute(ActivityStarter.java:742)
	at com.android.server.wm.ActivityTaskManagerService.startActivityAsUser(ActivityTaskManagerService.java:1290)
	at com.android.server.wm.ActivityTaskManagerService.startActivityAsUser(ActivityTaskManagerService.java:1241)

`
TraceLog of version 1.0.8
App crashes when opened and becomes unusuable, this also happened in older versions, but could resolve by storage cleaning in settings.
[image]
This is what it shows when I start it, started without a name and no, it does not work

simondankelmann, 2024-01-05:
Thank you for the Log. It seems your Bluetooth was disabled, can you try again with bluetooth turned on ?

Doubble29x, 2024-01-05:
Oh, mh, it seems that you need to have bluetooth started before setting the app for the first time. Seems to be working flawlessly now, thanks

Doubble29x, 2024-01-05:
```
type: crash
osVersion: google/raven/raven:14/UQ1A.240105.002/2024010400:user/release-keys
package: de.simon.dankelmann.bluetoothlespam:1
process: de.simon.dankelmann.bluetoothlespam
processUptime: 70 + 305 ms
installer: com.android.packageinstaller

java.lang.RuntimeException: Unable to create service de.simon.dankelmann.bluetoothlespam.Services.AdvertisementForegroundService: kotlin.UninitializedPropertyAccessException: lateinit property _activity has not been initialized
	at android.app.ActivityThread.handleCreateService(ActivityThread.java:4722)
	at android.app.ActivityThread.-$$Nest$mhandleCreateService(Unknown Source:0)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2296)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loopOnce(Looper.java:205)
	at android.os.Looper.loop(Looper.java:294)
	at android.app.ActivityThread.main(ActivityThread.java:8279)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
	at com.android.internal.os.ExecInit.main(ExecInit.java:49)
	at com.android.internal.os.RuntimeInit.nativeFinishInit(Native Method)
	at com.android.internal.os.RuntimeInit.main(RuntimeInit.java:359)
Caused by: kotlin.UninitializedPropertyAccessException: lateinit property _activity has not been initialized
	at de.simon.dankelmann.bluetoothlespam.AppContext.AppContext$Companion.getActivity(AppContext.kt:49)
	at de.simon.dankelmann.bluetoothlespam.Services.AdvertisementForegroundService.createNotificationChannel(AdvertisementForegroundService.kt:85)
	at de.simon.dankelmann.bluetoothlespam.Services.AdvertisementForegroundService.onCreate(AdvertisementForegroundService.kt:55)
	at android.app.ActivityThread.handleCreateService(ActivityThread.java:4709)
	... 11 more

```
oof, foreground crash now
Steps were:
Start iphone modal list
Get near homepod and boom crashed :0

Doubble29x, 2024-01-05:
Nvm, this foreground one had to do with bluetooth turning off after 15 seconds

simondankelmann, 2024-01-05:
I have recreated the APK's in the Release. can you please delete everything and try again ?

closed 2024-03-06 by Doubble29x