simondankelmann/Bluetooth-LE-Spam #219: App always crashing

Weirdrandomthings, 2024-01-06:
Both version 1.0.7 and 1.0.8 always crash when I try to open them, I have a rooted pixel 7a.
Deleting the app data and cache sadly does not work.
1.0.6 just works normally only those two versions don't.

simondankelmann, 2024-01-10:
1.0.7 and 1.0.8 need the database to be recreated. it seems some android device do not clear the data correctly. if you tried the debug version please get release version, if not: please try the debug verison. if the problem persists, please try debug version again and provide some logs so we can find out what is happening :)

Weirdrandomthings, 2024-01-10:
Sorry but I don't know how to find the logs.

Weirdrandomthings, 2024-01-17:
I finally got it running, I used titanium backup to delete all app data and then I deleted the app and reinstalled it, now it works👍

closed 2024-01-17 by Weirdrandomthings

simondankelmann, 2024-01-18:
Nice ! Thanks for sharing a solution!