simondankelmann/Bluetooth-LE-Spam #222: Friends phone keeps on crashing after spamming him for a minute.

CinnamonRob, 2024-01-08:
I did this to my friend as a joke and now if his phone is not plugged in or connected to WiFi then after 10 seconds of the screen being on it will crash and be unresponsive for a minute.

simondankelmann, 2024-01-10:
what kind of phone do you and your friend have, is this issue still present ?

user88294, 2024-02-02:
Crashing certain IOS devices is an intended feature that's part of the "Continuity" collection, it is inside of the "IOS 17 Crash List" and when it runs it can crash IOS devices without problem. If you don't wish for this to happen, use filters to select another list or just restart it if it's close to the beginning of the list.

Prololgh2, 2024-02-06:
i have the opposite problem me and my friend were messing with this i have a s23 ultra he has a 14 and before that a se and we couldn't get it to crash even tho we did as the respiratory any tips?