simondankelmann/Bluetooth-LE-Spam #224: Crashing on start

Eohgsuiq, 2024-01-10:
I have stock redmi note 10s 
Here is the log:
java.lang.IllegalArgumentException: Can't convert value to enum, unknown value: ADVERTISEMENT_TYPE_CONTINUITY_DEVICE_POPUPS
	at de.simon.dankelmann.bluetoothlespam.Database.Dao.AdvertisementSetDao_Impl.__AdvertisementSetType_stringToEnum(AdvertisementSetDao_Impl.java:605)
	at de.simon.dankelmann.bluetoothlespam.Database.Dao.AdvertisementSetDao_Impl.getAll(AdvertisementSetDao_Impl.java:245)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment$checkDatabase$1.invokeSuspend(StartFragment.kt:514)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at kotlinx.coroutines.internal.LimitedDispatcher.run(LimitedDispatcher.kt:42)
	at kotlinx.coroutines.scheduling.TaskImpl.run(Tasks.kt:95)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:570)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:677)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:664)
	Suppressed: kotlinx.coroutines.DiagnosticCoroutineContextException: [StandaloneCoroutine{Cancelling}@888aeac, Dispatchers.IO]

simondankelmann, 2024-01-10:
This log shows that the old Database still exists, please make sure to delete all application data and cache, then try again.

simondankelmann, 2024-01-23:
Closed as there was no further information provided

closed 2024-01-23 by simondankelmann