simondankelmann/Bluetooth-LE-Spam #227: Not working

susnox, 2024-01-16:
So i've tried every version and the only thing that works is IOS AppleTV related things, android nor windows aren't working for me

stepanzak, 2024-01-18:
> So i've tried every version and the only thing that works is IOS AppleTV related things, android nor windows aren't working for me

Android works for only a few devices for me.
iOS modals and crash list works relatively well, other iOS lists works only when I basically touch the iPhone with my phone. Windows stopped working for me. Friend of mine has a phone with modern bluetooth support, and it works much better than mine that says only legacy bluetooth support. You can check bluetooth support on the Bluetooth LE Spam's main screen.

simondankelmann, 2024-01-19:
Yes, sadly the range of most payloads is very low. apple action modals and ios crash list (which is basically a modified action modals list) work in a way that allow the sender and receiver being in a higher distance to each other. for apple device popups and android fast pairing popups the receiver calculates the proximity and only shows a popup when it thinks the device is in very close range.

closed 2024-01-19 by simondankelmann