simondankelmann/Bluetooth-LE-Spam #229: Working on Android 8.0 and later

Fusseldieb, 2024-01-21:
I have an S10e with Android 13 and the App works on iOS devices just fine.

Reporting this because README states that it would only work properlyon devices older than Android 8.0.

Vniuo, 2024-01-22:
Why Xiaomi's not responding

Fusseldieb, 2024-01-22:
> Why Xiaomi's not responding

What do you mean not responding?

Vniuo, 2024-01-22:
> > Why Xiaomi's not responding
> 
> What do you mean not responding?

It's about choosing the same device as Xiaomi's Bluetooth, but when sending it out, Xiaomi's phone doesn't respond (I don't know English, so I used a translator, which may have deviated from your communication)

simondankelmann, 2024-01-23:
"Android 8.0 (API level 26) or later does not support IOS or PC" is maybe  a bit missleading. it means the app is made for android 8 and above and you cannot install it on your Windows PC or iOs Device.

closed 2024-01-23 by simondankelmann

Vniuo, 2024-01-23:
I know, my device is Xiaomi, and the Android is 14