simondankelmann/Bluetooth-LE-Spam #231: Crashing at start on every oppo device

SFYMMIK, 2024-01-22:
the latest version crashes as soon as i open it on any oppo device idk why

simondankelmann, 2024-01-22:
Please try to delete all App data and cache in your phone's app settings

SFYMMIK, 2024-01-22:
> Please try to delete all App data and cache in your phone's app settings

it still does not work

simondankelmann, 2024-01-23:
Can you provide any log data ?

SFYMMIK, 2024-01-25:
> Can you provide any log data ?

idk how to do that

SteadyStatus21, 2024-01-25:
> > Please try to delete all App data and cache in your phone's app settings
> 
> it still does not work

Could you try the following: 

Ensure the Location Permission is set to "Allow All the Time", along with using "Precise Location"? That solved this issue for me. 

Follow up, @simondankelmann why does the app always need permissions to my fine location? Should I make a separate issue to discuss this?

simondankelmann, 2024-01-25:
We need that for the spam detection feature. its used for bluetooth le scanning in the background

SteadyStatus21, 2024-01-27:
> We need that for the spam detection feature. its used for bluetooth le scanning in the background

Thank you for the response, and that makes a lot of sense now! Thank you Simon! Keep up the great work with this app!

closed 2025-01-23 by ha2ked