simondankelmann/Bluetooth-LE-Spam #235: Crash on start if some permission is not granted

licaon-kter, 2024-01-26:
1.0.8 F-Droid
Lineage based Android 13

Log:
```
FATAL EXCEPTION: main
Process: de.simon.dankelmann.bluetoothlespam, PID: 19634
java.lang.StackOverflowError: stack size 8188KB
	at android.permission.PermissionManager$PermissionQuery.hashCode(PermissionManager.java:1568)
	at java.util.HashMap.hash(HashMap.java:338)
	at java.util.LinkedHashMap.get(LinkedHashMap.java:464)
	at android.app.PropertyInvalidatedCache.query(PropertyInvalidatedCache.java:939)
	at android.permission.PermissionManager.checkPermission(PermissionManager.java:1605)
	at android.app.ContextImpl.checkPermission(ContextImpl.java:2175)
	at android.content.ContextWrapper.checkPermission(ContextWrapper.java:910)
	at androidx.core.content.ContextCompat.checkSelfPermission(ContextCompat.java:601)
	at de.simon.dankelmann.bluetoothlespam.PermissionCheck.PermissionCheck$Companion.checkPermission(PermissionCheck.kt:60)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment.checkRequiredPermissions(StartFragment.kt:586)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment._init_$lambda$9(StartFragment.kt:629)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment.$r8$lambda$pxmCLAyLfTMjqqQSnXGEgbFmthM(Unknown Source:0)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment$$ExternalSyntheticLambda2.onActivityResult(Unknown Source:4)
	at androidx.activity.result.ActivityResultRegistry.doDispatch(ActivityResultRegistry.java:414)
	at androidx.activity.result.ActivityResultRegistry.dispatchResult(ActivityResultRegistry.java:371)
	at androidx.activity.ComponentActivity.onRequestPermissionsResult(ComponentActivity.java:870)
	at androidx.fragment.app.FragmentActivity.onRequestPermissionsResult(FragmentActivity.java:489)
	at android.app.Activity.requestPermissions(Activity.java:5397)
	at androidx.core.app.ActivityCompat$Api23Impl.requestPermissions(ActivityCompat.java:893)
	at androidx.core.app.ActivityCompat.requestPermissions(ActivityCompat.java:557)
	at androidx.activity.ComponentActivity$1.onLaunch(ComponentActivity.java:218)
	at androidx.activity.result.ActivityResultRegistry$2.launch(ActivityResultRegistry.java:173)
	at androidx.fragment.app.Fragment$10.launch(Fragment.java:3632)
	at androidx.activity.result.ActivityResultLauncher.launch(ActivityResultLauncher.java:47)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment.checkBackgroundLocationAccessPermission(StartFragment.kt:620)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment.checkRequiredPermissions(StartFragment.kt:604)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment._init_$lambda$9(StartFragment.kt:629)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment.$r8$lambda$pxmCLAyLfTMjqqQSnXGEgbFmthM(Unknown Source:0)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment$$ExternalSyntheticLambda2.onActivityResult(Unknown Source:4)
	at androidx.activity.result.ActivityResultRegistry.doDispatch(ActivityResultRegistry.java:414)
	at androidx.activity.result.ActivityResultRegistry.dispatchResult(ActivityResultRegistry.java:371)
	at androidx.activity.ComponentActivity.onRequestPermissionsResult(ComponentActivity.java:870)
	at androidx.fragment.app.FragmentActivity.onRequestPermissionsResult(FragmentActivity.java:489)
	at android.app.Activity.requestPermissions(Activity.java:5397)
	at androidx.core.app.ActivityCompat$Api23Impl.requestPermissions(ActivityCompat.java:893)
	at androidx.core.app.ActivityCompat.requestPermissions(ActivityCompat.java:557)
	at androidx.activity.ComponentActivity$1.onLaunch(ComponentActivity.java:218)
	at androidx.activity.result.ActivityResultRegistry$2.launch(ActivityResultRegistry.java:173)
	at androidx.fragment.app.Fragment$10.launch(Fragment.java:3632)
	at androidx.activity.result.ActivityResultLauncher.launch(ActivityResultLauncher.java:47)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment.checkBackgroundLocationAccessPermission(StartFragment.kt:620)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment.checkRequiredPermissions(StartFragment.kt:604)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment._init_$lambda$9(StartFragment.kt:629)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment.$r8$lambda$pxmCLAyLfTMjqqQSnXGEgbFmthM(Unknown Source:0)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment$$ExternalSyntheticLambda2.onActivityResult(Unknown Source:4)
	at androidx.activity.result.ActivityResultRegistry.doDispatch(ActivityResultRegistry.java:414)
	at androidx.activity.result.ActivityResultRegistry.dispatchResult(ActivityResultRegistry.java:371)
	at androidx.activity.ComponentActivity.onRequestPermissionsResult(ComponentActivity.java:870)
	at androidx.fragment.app.FragmentActivity.onRequestPermissionsResult(FragmentActivity.java:489)
	at android.app.Activity.requestPermissions(Activity.java:5397)
	at androidx.core.app.ActivityCompat$Api23Impl.requestPermissions(ActivityCompat.java:893)
	at androidx.core.app.ActivityCompat.requestPermissions(ActivityCompat.java:557)
	at androidx.activity.ComponentActivity$1.onLaunch(ComponentActivity.java:218)
	at androidx.activity.result.ActivityResultRegistry$2.launch(ActivityResultRegistry.java:173)
	at androidx.fragment.app.Fragment$10.launch(Fragment.java:3632)
	at androidx.activity.result.ActivityResultLauncher.launch(ActivityResultLauncher.java:47)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment.checkBackgroundLocationAccessPermission(StartFragment.kt:620)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment.checkRequiredPermissions(StartFragment.kt:604)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment._init_$lambda$9(StartFragment.kt:629)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment.$r8$lambda$pxmCLAyLfTMjqqQSnXGEgbFmthM(Unknown Source:0)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment$$ExternalSyntheticLambda2.onActivityResult(Unknown Source:4)
	at androidx.activity.result.ActivityResultRegistry.doDispatch(ActivityResultRegistry.java:414)
[... 125 lines cut ...]
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment.$r8$lambda$pxmCLAyLfTMjqqQSnXGEgbFmthM(Unknown Source:0)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment$$ExternalSyntheticLambda2.onActivityResult(Unknown Source:4)
	at androidx.activity.result.ActivityResultRegistry.doDispatch(ActivityResultRegistry.java:414)
	at androidx.activity.result.ActivityResultRegistry.dispatchResult(ActivityResultRegistry.java:371)
	at androidx.activity.ComponentActivity.onRequestPermissionsResult(ComponentActivity.java:870)
	at androidx.fragment.app.FragmentActivity.onRequestPermissionsResult(FragmentActivity.java:489)
	at android.app.Activity.requestPermissions(Activity.java:5397)
	at androidx.core.app.ActivityCompat$Api23Impl.requestPermissions(ActivityCompat.java:893)
	at androidx.core.app.ActivityCompat.requestPermissions(ActivityCompat.java:557)
	at androidx.activity.ComponentActivity$1.onLaunch(ComponentActivity.java:218)
	at androidx.activity.result.ActivityResultRegistry$2.launch(ActivityResultRegistry.java:173)
	at androidx.fragment.app.Fragment$10.launch(Fragment.java:3632)
	at androidx.activity.result.ActivityResultLauncher.launch(ActivityResultLauncher.java:47)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment.checkBackgroundLocationAccessPermission(StartFragment.kt:620)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment.checkRequiredPermissions(StartFragment.kt:604)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment._init_$lambda$9(StartFragment.kt:629)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment.$r8$lambda$pxmCLAyLfTMjqqQSnXGEgbFmthM(Unknown Source:0)
	at de.simon.dankelmann.bluetoothlespam.ui.start.StartFragment$$ExternalSyntheticLambda2.onActivityResult(Unknown Source:4)
	at androidx.activity.result.ActivityResultRegistry.doDispatch(ActivityResultRegistry.java:414)
	at androidx.activity.result.ActivityResultRegistry.dispatchResult(ActivityResultRegistry.java:371)
```

While trying to give it the least amount of perms, maybe the app should try/catch such exceptions and retrigger the perms screens as needed.

Also maybe explain which level of perms is wanted because I have like 3 options for Location, and to be fair I should be able to use "Ask every time" as expected.

simondankelmann, 2024-01-28:
Thanks for sharing the Log, this will be helpful for future enhancements :)

DebugBoard, 2025-02-25:
Its on my list of things to fix. So soon™

closed 2025-03-10 by DebugBoard