simondankelmann/Bluetooth-LE-Spam #245: Dosent work

Zayedbj, 2024-02-27:
I installed the app and then I tried the Bluetooth spam and it dosent work,I am currently using samsung A52 and I have an xiamoi tablet and a xiamoi phone 11 lite 5g and I tried to spam them (the Bluetooth was on) and it didn't work. Can someone help??

Denalgus2, 2024-02-29:
It looks like most Android phones have patched this

ha2ked, 2024-02-29:
There not much we can do sorry 😔.

simondankelmann, 2024-03-01:
i too had the feeling that much less popups appear nowadays. maybe the google devs implemented some limitations ?!

Zayedbj, 2024-03-01:
So I tried it again and it worked with iPhones but it dosent keep spamming it does the Pop on the iPhone like every 30 sec or so

simondankelmann, 2024-03-01:
Is the iPhone updated to the latest iOS ? This really sounds like they limited the number of popups

simondankelmann, 2024-03-01:
I assume you tried the action modals ?

Zayedbj, 2024-03-01:
> Is the iPhone updated to the latest iOS ? This really sounds like they limited the number of popups

Yes it is I think they really limited the Pop ups

Zayedbj, 2024-03-01:
> I assume you tried the action modals ?

Action modals? Wdym?

simondankelmann, 2024-03-01:
I mean when you spammed the iPhone, did you go to the continuity page and selected the iOS action modals list ?

Denalgus2, 2024-03-01:
The ios 17.2 update patched most of the spamming, it is no longer possible to crash the iPhones, and the popups are much more limited. For the androids I've noticed that it works really well on the older phones, while the newest ones are not affected by the spam at all

simondankelmann, 2024-03-01:
At least this whole topic brought so much attention that it had to be fixed 😅

prololgh1, 2024-03-01:
On my s23 ultra it works well even on 17.4 I cant crash my friends phone but I don't mind that. I also crashed a app lol on 17.4 btw he's the one that showed me this

closed 2024-03-14 by simondankelmann