simondankelmann/Bluetooth-LE-Spam #246: Location permission allow only while using the app

Mark7888, 2024-02-29:
Why does this app need the location permission set to `always`? It doesn't even start, if I change the location permission to `allow only while using the app`. What's the reason it demands the `always` permission?

I generally don't really like the idea of granting constant locaiton permission to an application. There are some exceptions, like google maps, but in my opinion, this application supposed to work just fine with the other permission setting.

ha2ked, 2024-02-29:
It's for the spam detector so that we can detect  in real time and get a notification. It doesn't need it if you don't want that feature. We don't collect any data anyway.

simondankelmann, 2024-03-01:
Then we might have a bug here. As @ha2ked already explained, the benefit of using this permission is to continue ble scanning in the background and notify on any detected spam or flipper devices. it should still be possible to use the app without granting, youre right there

yashikada, 2024-05-19:
> it should still be possible to use the app without granting, youre right there

For me no, I have a Samsung A72 with OneuI 6.0 if I don't set always allow, the app continue to popup the required permission. I think you should accept "ask every time", or "while the app is in use".

yashikada, 2024-05-19:
I add also from Android 12 Location permission can be omitted:

[Here](https://source.android.com/docs/core/connect/bluetooth/ble) a details.
```
Location scanning

A device's location services can use Bluetooth to detect Bluetooth beacons and provide a more accurate location. This is how users enable the feature:

    Go to Settings > Security & Location > Location.
    Check that the Location On/Off toggle is in the On position.
    Go to Mode.
    Select either High accuracy or Battery saving mode.

In Android 11 or lower, individual apps require location permissions to use BLE scanning, even if they are scanning only to find devices to connect to. If the user disables location scanning, or does not grant an app location permissions, then the app won't receive any BLE scanning results.

From Android 12, the[ BLUETOOTH_SCAN](https://developer.android.com/reference/android/Manifest.permission#BLUETOOTH_SCAN), BLUETOOTH_ADVERTISE, and[ BLUETOOTH_CONNECT](https://developer.android.com/reference/android/Manifest.permission#BLUETOOTH_CONNECT) permissions can allow apps to scan for nearby devices without needing to request the location permission. For more information, see[ New Bluetooth permissions in Android 12](https://developer.android.com/about/versions/12/features/bluetooth-permissions).

Users can disable system-level Bluetooth background scanning by going to Settings > Security & Location > Location > Scanning and disabling the toggle for Bluetooth scanning. This does not affect BLE scanning for location or local devices.
```

@simondankelmann I hope you can improve it.

v1ctorio, 2024-11-25:
[image]
The app won't even start if not granted the ACCESS_BACKGROUND_LOCATION permission. This shouldn't be like this, or is this somehow a requerinent for BLE or anything?

ha2ked, 2024-12-25:
This is an intended feature therefore not an issue.

closed 2024-12-25 by ha2ked

v1ctorio, 2024-12-25:
> This is an intended feature therefore not an issue.

Can you elaborate please? Why is it not possible to use the app at all without the **background** location permission? 

> It doesn't need it if you don't want that feature.

That does not happen, you do need to give the permission to use the app

yashikada, 2024-12-26:
> Can you elaborate please?

there is nothing to elaborate since it doesn't know how to solve the issue so it says it's a feature, like apple developers do. I stopped using this useless app.