simondankelmann/Bluetooth-LE-Spam #247: Not working on Redmi Note 10 4G variant...

marksuhajda94, 2024-02-29:
I saw a video about your app but it does not work on my phone. I have a Xiaomi Redmi Note 10 4G variant phone with Android 12. The app installed correctly, it has all of the permissions set and when i tap an advertisement and hit the play button it tried to play the advertisement but error messages kept showing up (ADVERTISE_FAILED_UNKNOWN).
[image]

ha2ked, 2024-02-29:
Log pls

marksuhajda94, 2024-02-29:
Here is the log file, I used the Fast Pair Collection for testing.

[Redmi_note_10_bluetoothlespam_logcat.txt](https://github.com/simondankelmann/Bluetooth-LE-Spam/files/14452696/Redmi_note_10_bluetoothlespam_logcat.txt)

simondankelmann, 2024-03-01:
On the Startscreen, does it say: "Bluetooth: Modern & Legacy" ? if yes, please go to Settings, turn off Legacy Advertising and restart the App. If that does not help, please try to delete all app data and cache and restart the app, once restarted please remain on the startscreen until all red bars are gone. especially the one for the database seeding. maybe something got corrupted there.

marksuhajda94, 2024-03-01:
I switched off the legacy mode, it was on previously but there is no change, it remains legacy only on the start screen.

marksuhajda94, 2024-03-01:
After deleting cache and app data, it still does not work.

simondankelmann, 2024-03-14:
Ok so your device only supports legacy advetising please turn it back on. The log says that a file from app data could not be loaded. Could you try to delete all App data and cache, then remove the app completely and install the release version instead of debug ? Please also manually check if all permissions are granted

marksuhajda94, 2024-03-23:
I explain what I did. I deleted both the release and debug version. I reinstalled the release version and before I opened the app, I checked every permission and manually set them under the app settings on the phone and then I opened up the app and now it works like a charm... Thanks for helping me out!

closed 2024-03-23 by marksuhajda94