simondankelmann/Bluetooth-LE-Spam #254: Translation

mikropsoft, 2024-03-14:
Is the app available for language translation? I will open a pr

simondankelmann, 2024-03-15:
hi :) what exactly do you mean ? right now i didnt plan to translate it. how would you like to do it ?

mikropsoft, 2024-03-15:
> hi :) what exactly do you mean ? right now i didnt plan to translate it. how would you like to do it ?

Strings.xml file is available. If you allow me I would like to translate it into my language. I will open a pull request. The reason I ask this is that the application is a single language.

simondankelmann, 2024-03-15:
Will this translate automatically? I never worked with that before 😅 of course I would appreciate your input, I just want to make sure that English remains as fallback language for all untranslated locales

mikropsoft, 2024-03-15:
> Will this translate automatically? I never worked with that before 😅 of course I would appreciate your input, I just want to make sure that English remains as fallback language for all untranslated locales

Then I will open pr today if it is convenient for you

closed 2024-03-15 by mikropsoft