simondankelmann/Bluetooth-LE-Spam #256: IOS 17 Popups are really slow

PatrikPepega, 2024-03-15:
Hello, i tried spamming my friend iPhones (12 Mini and 15 Pro Max), but all of the popups take around minute or 30seconds to show and in most cases it won't even show up on iphone. I tried changing the milliseconds, but its still same. I also tried spamming Samsung phone, and it was way faster. Upon closing a popup it instantly showed another one. How can i do the same thing to iPhones?

EinJonas187, 2024-03-15:
Try to set the interval below 100ms

Alluseri, 2024-03-21:
Same here, interval makes no difference, 1200ms is even more reliable.

PatrikPepega, 2024-04-06:
> Try to set the interval below 100ms

Still same nothing changed.

mmwong1984, 2024-04-29:
any solution?

simondankelmann, 2024-04-29:
this seems to be patched by apple. popups on latest ios versions do appear limited as i understood

closed 2024-04-29 by simondankelmann